PX4/PX4-Autopilot #13595: HITL fixedwing and VTOL in fixedwing mode not working

julianoes, 2019-11-26:
**Describe the bug**
It seems like HITL with fixedwing or VTOL in fixedwing is not working. Presumably, the actuator setpoints are not set/connected correctly.

**To Reproduce**
Steps to reproduce the behavior with fixedwing:
1. Connect Pixhawk over USB.
2. Set `SYS_AUTOSTART` to 2100 for fixedwing, reboot Pixhawk.
3. Edit `Tools/sitl_gazebo/models/plane/plane.sdf` and set `serialEnabled` and `hil_mode` to `true`.
4. Start Gazebo:
   ```
   source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
   gazebo --verbose Tools/sitl_gazebo/worlds/plane.world
   ```
5. Try flying using joysticks in QGC.

**Expected behavior**
I expect that I can fly the plane in Stabilized, and not crash.

julianoes, 2019-11-26:
For fixedwing I found that it is a mixer issue. We'll need a specific mixer for the Gazebo HITL case.

rahuldeo2047, 2023-03-07:
Fixed wing.

Running via the documented make command for version v1.14.alpha.1 . That works for mission mode perfectly.

issue was
I was working with v1.12.3 (for Xplan, I guess have to shift to other sims) and running gazebo with v1.14.alpha.1 plane world. I did some temporary changes in the mixed files for my RC to work. The mission mode works, and manual mode doesn’t work. In the mission mode after launch and takeoff the plan behaves lost control and falls. 

(Unrelated to the issue: My RC did not work with both the versions. Debugging)

rahuldeo2047, 2023-03-07:
However, the gazebo and plane world did not work. Still debugging. I will confirm.