PX4/PX4-Autopilot #13607: version problem when working with AirSim

xlong0513, 2019-11-27:
Hi, i'm using PX4 with AirSim for SITL, and version 1.8.2 is recommended. The code is as follow
`make posix_sitl_ekf2  none_iris`
But when checkout to `master` branch, this command cannot be used.
What's the new command to replace it with the master branch?
Thanks.

julianoes, 2019-11-27:
Can you give some more context on what commands you are running? And what the output is and what the problem is, etc.

hamishwillee, 2019-11-27:
@julianoes They will be following these instructions to start SITL before starting airsim: https://github.com/microsoft/AirSim/blob/master/docs/px4_sitl.md

xlong0513, 2019-11-28:
@julianoes Sorry for too little description. just as @hamishwillee  supplemented. 
I mean I cannot find this command `make posix_sitl_ekf2  none_iris` in master branch. Does it change?

julianoes, 2019-11-28:
Oh, now I understand. The command is now:

```
make px4_sitl none_iris
```

However, note that AirSim does not work with master or 1.9.2 yet unless you [disable lockstep](https://dev.px4.io/v1.9.0/en/simulation/index.html#disable-lockstep-simulation).
It's [in the works](https://github.com/microsoft/AirSim/pull/2310) though.

xlong0513, 2019-11-29:
@julianoes Hi, thanks for the reply. i have tried the link you recommend. In detail, i have disabled the lockstep by `set(ENABLE_LOCKSTEP_SCHEDULER no)`, and modified the setting file. The following is my setting.json file.
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
        "VehicleType": "PX4Multirotor",
        "UseSerial": false,
        "UseTcp": true,
        "TcpPort": 4560,
        "GroundControlPort": 14580,
        "Parameters": {
            "NAV_RCL_ACT": 0,
            "NAV_DLL_ACT": 0,
            "LPE_LAT": 47.641468,
            "LPE_LON": -122.140165,
            "COM_OBL_ACT": 1
        }
    }
  }
}
`
However, it still doesn't work. Maybe 1.8.2 is more stable. Besides, the AirSim always crashed when i use "PX4" as the "Vehicles". Can u tell us what the problem is?

xlong0513, 2019-11-29:
@julianoes  Hi, thanks for the reply. i have tried the link you recommend. In detail, i have disabled the lockstep by set(ENABLE_LOCKSTEP_SCHEDULER no), and modified the setting file. The following is my setting.json file.
[image]

However, it still doesn't work. Maybe 1.8.2 is more stable. Besides, the AirSim always crashed when i use "PX4" as the "Vehicles". Can u tell us what the problem is?

mhkabir, 2019-12-05:
@lovettchris Can you help here?

julianoes, 2019-12-06:
@xlong0513 try with Udp port 15460 instead of TCP.

And replace this line:
https://github.com/PX4/Firmware/blob/993fa5bd3758030e6b461109f4b0ae8da9cec388/ROMFS/px4fmu_common/init.d-posix/rcS#L234

with: 
```
simulator start -u 14560
```

chris-aeviator, 2019-12-07:
I'm successfully running px4 (v 1.8.2) SITL with QGroundcontrol & Airsim

My build & run command is 
```
make px4_sitl_default none_iris
sudo ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
```
[/AirSim/settings.json]
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UdpPort": 14560
        }
    }
}

```

julianoes, 2020-03-06:
Airsim is now compatible with lockstep, closing.

closed 2020-03-06 by julianoes