PX4/PX4-Autopilot #13609: Here2 GPS/mag over UAVCAN not working

julianoes, 2019-11-27:
**Describe the bug**
It seems like the GPS and mag of the Here2 GPS via UAVCAN are not working.

**To Reproduce**
1. Update Here2 GPS using MissionPlanner to latest and greatest.
2. Connect Here2 with Pixhawk 4 to Can1 port.
3. Set `UAVCAN_ENABLE` to 2 (Sensors automatic config).
4. Check `listener sensor_mag`, check `gps status`. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**

```
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [uavcan] sensor bridge 'baro' init ok
INFO  [uavcan] sensor bridge 'mag' init ok
INFO  [uavcan] sensor bridge 'gnss' init ok
INFO  [uavcan] sensor bridge 'flow' init ok
INFO  [uavcan] sensor bridge 'battery' init ok
ERROR [uavcan] CentralizedServer init: -1
ERROR [uavcan] Node init failed: -1
ERROR [uavcan] Firmware Server Failed to Start -1
```

```
nsh> uavcan status
Pool allocator status:
        Capacity hard/soft: 500/250 blocks
        Reserved:  2 blocks
        Allocated: 2 blocks

UAVCAN node status:
        Internal failures: 0
        Transfer errors:   0
        RX transfers:      0
        TX transfers:      85

CAN1 status:
        HW errors: 182
        IO errors: 182
        RX frames: 35
        TX frames: 57
CAN2 status:
        HW errors: 8939
        IO errors: 8939
        RX frames: 0
        TX frames: 57

control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 0
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 8: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 9: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 10: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 11: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 12: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 13: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 14: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 15: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191

Sensor 'baro':
devname: /dev/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'mag':
devname: /dev/mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'gnss':
RX errors: 0, using old Fix: 1, receiver node id: N/A

Sensor 'flow':
devname: /dev/flow
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'battery':
devname: /dev/battery
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 9619 events, 225719us elapsed, 23.47us avg, min 11us max 971us 24.913us rms
uavcan: cycle interval: 9619 events, 2955.38us avg, min 25us max 4276us 381.113us rms

```

```
nsh> listener sensor_mag

TOPIC: sensor_mag
 sensor_mag_s
        timestamp: 96218552  (0.013206 seconds ago)
        error_count: 0
        device_id: 396825 (Type: 0x06, I2C:3 (0x0E)) 
        x: 0.0630
        y: -0.2040
        z: 0.3090
        temperature: 0.0000
        scaling: 0.0030
        x_raw: 21
        y_raw: -68
        z_raw: 103
        is_external: False
nsh> listener vehicle_gps_position
never published
nsh> 
```

**Additional context**
Version: 09c8c8f706c5f6df3d42ee968f9c9d52cbf2e25a

@pavel-kirienko what am I doing wrong? And where do I find more docs about this?

dagar, 2019-11-27:
Do you have a babel or something you can use to see what's on the CAN bus?

julianoes, 2019-11-27:
@dagar maybe, I'd have to check.

ayberkozgur, 2020-02-07:
I have the same problem, there is nothing but the here2 and babel `NodeStatus` on the bus (pixhawk not on the bus).
[image]

hichamhendy, 2022-11-30:
Any news on that?