PX4/PX4-Autopilot #13612: Commander class structure instead of scattered static states

MaEtUgR, 2019-11-27:
**Describe problem solved by the proposed feature**
I'm trying to split out all the preflight/arming logic of the commander into modular classes with the main goal to have them more understandable and callable in a modular way on a regular basis. https://github.com/PX4/Firmware/pull/13224 was the first step of this but now **I'm running against a wall adding member variables to the classes and having the class instances be part of the commander module class** since there are so many calls from outside static functions.

One chunk of Commander improvements also listed in #7055 is to have the Commander module as a class holding all the split out subfunctions instead of having all the remaining static states and functions e.g. here https://github.com/PX4/Firmware/blob/214e9c8244bfca72a319694ae222727f658ca6c1/src/modules/commander/Commander.cpp#L112-L270

**Describe your preferred solution**
I think a really big point that is missing is **getting rid of the low priority thread** for calibration and a solution moving it to the events was last tracked here: https://github.com/PX4/Firmware/pull/8706 (older PRs for the same thing #8200 #8630) and after that I think it's a lot easier to achieve this goal step by step because the low priority thread is in my eyes the blocker for refactoring since it's more complicated from a scheduling perspective.

I plan on trying to take this up again and do progress. Can we collect the main helping pointers and concerns here such that it gets easy to take a stab?

FYI @dagar

MaEtUgR, 2019-11-28:
Update: @dagar Already opened a pr yesterday to work into that direction: https://github.com/PX4/Firmware/pull/13613 (Thanks for that!)

bkueng, 2022-10-25:
All done

closed 2022-10-25 by bkueng