PX4/PX4-Autopilot #13616: Undefined reference to PX4Rangefinder adding tfmini in px4_fmu-v2_default make

gianni-carbone, 2019-11-27:
error message then stop compiling when try to add tfmini to default make of px4_fmu-v2_default.

**default.cmake:**
px4_add_board(
	PLATFORM nuttx
	VENDOR px4
	MODEL fmu-v2
	LABEL default
	TOOLCHAIN arm-none-eabi
	ARCHITECTURE cortex-m4
	ROMFSROOT px4fmu_common
	BOOTLOADER ${PX4_SOURCE_DIR}/ROMFS/px4fmu_common/extras/px4fmuv3_bl.bin
	IO px4_io-v2_default
	#TESTING
	#UAVCAN_INTERFACES 2
	CONSTRAINED_FLASH
	SERIAL_PORTS
		GPS1:/dev/ttyS3
		TEL1:/dev/ttyS1
		TEL2:/dev/ttyS2
		TEL4:/dev/ttyS6
	DRIVERS
		adc
		#barometer # all available barometer drivers
		barometer/ms5611
		#batt_smbus
		#camera_capture
		#camera_trigger
		#differential_pressure # all available differential pressure drivers
		#differential_pressure/ms4525
		#distance_sensor # all available distance sensor drivers
		#distance_sensor/ll40ls
		#distance_sensor/sf0x
		distance_sensor/tfmini
		#dshot
		gps
		#heater
		#imu # all available imu drivers
		#imu/adis16448
		#imu/adis16477
		#imu/adis16497
		imu/l3gd20
		imu/lsm303d
		imu/mpu6000
		#imu/mpu9250
		#iridiumsbd
		#irlock
		#lights/blinkm
		lights/rgbled
		#magnetometer # all available magnetometer drivers
		magnetometer/hmc5883
		#mkblctrl
		#optical_flow # all available optical flow drivers
		#optical_flow/px4flow
		#osd
		#pca9685
		#protocol_splitter
		#pwm_input
		pwm_out_sim
		px4fmu
		px4io
		#roboclaw
		#tap_esc
		#telemetry # all available telemetry drivers
		#test_ppm
		tone_alarm
		#uavcan
	MODULES
		#airspeed_selector
		#attitude_estimator_q
		battery_status
		camera_feedback
		commander
		dataman
		ekf2
		#events
		#fw_att_control
		#fw_pos_control_l1
		land_detector
		#landing_target_estimator
		load_mon
		#local_position_estimator
		logger
		mavlink
		mc_att_control
		mc_pos_control
		mc_rate_control
		navigator
		rc_update
		#rover_pos_control
		sensors
		#sih
		vmount
		#vtol_att_control
	SYSTEMCMDS
		bl_update
		#config
		#dmesg
		#dumpfile
		#esc_calib
		hardfault_log
		#i2cdetect
		#led_control
		mixer
		#motor_ramp
		#motor_test
		mtd
		#nshterm
		param
		perf
		pwm
		reboot
		#reflect
		#sd_bench
		#shutdown
		#tests # tests and test runner
		top
		#topic_listener
		tune_control
		#usb_connected
		ver
		#work_queue
	EXAMPLES
		#bottle_drop # OBC challenge
		#fixedwing_control # Tutorial code from https://px4.io/dev/example_fixedwing_control
		#hello
		#hwtest # Hardware test
		#matlab_csv_serial
		#px4_mavlink_debug # Tutorial code from http://dev.px4.io/en/debug/debug_values.html
		#px4_simple_app # Tutorial code from http://dev.px4.io/en/apps/hello_sky.html
		#rover_steering_control # Rover example app
		#uuv_example_app
	)

**building result:**
src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a(TFMINI.cpp.obj): In function `TFMINI::TFMINI(char const*, unsigned char)':
/home/gianni/src/20191127_230644/Firmware/build/px4_fmu-v2_default/../../src/drivers/distance_sensor/tfmini/TFMINI.cpp:38: undefined reference to `PX4Rangefinder::PX4Rangefinder(unsigned int, unsigned char, unsigned char)'
src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a(TFMINI.cpp.obj): In function `TFMINI::collect()':
/home/gianni/src/20191127_230644/Firmware/build/px4_fmu-v2_default/../../src/drivers/distance_sensor/tfmini/TFMINI.cpp:216: undefined reference to `PX4Rangefinder::update(unsigned long long, float, signed char)'
src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a(TFMINI.cpp.obj): In function `TFMINI::print_info()':
/home/gianni/src/20191127_230644/Firmware/build/px4_fmu-v2_default/../../src/drivers/distance_sensor/tfmini/TFMINI.cpp:261: undefined reference to `PX4Rangefinder::print_status()'
src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a(TFMINI.cpp.obj): In function `TFMINI::~TFMINI()':
/home/gianni/src/20191127_230644/Firmware/build/px4_fmu-v2_default/../../src/drivers/distance_sensor/tfmini/TFMINI.cpp:47: undefined reference to `PX4Rangefinder::~PX4Rangefinder()'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_fmu-v2_default' failed
make: *** [px4_fmu-v2_default] Error 1

mcsauder, 2019-11-28:
Thanks for reporting @gianni-carbone , please let us know if PR #13617 solves your issue!

mcsauder, 2019-11-28:
Closing with PR #13617!

closed 2019-11-28 by mcsauder