PX4/PX4-Autopilot #13631: [Feature Request] Support UP-Flow LC302 Optical Flow Module

bys1123, 2019-11-29:
UP-Flow LC302 is a very popular optical flow module in china, lots of University are using this module for indoor usage. And toys also use this chip a lot.

This also have better performance than PMW3901.

[image]
[image]

Purchase: aliexpress.com/item/32968961948.html?spm=a2g0o.productlist.0.0.4db0495eSoA2ss&algo_pvid=aadf0f14-60af-47be-b95a-0ff85d514030&algo_expid=aadf0f14-60af-47be-b95a-0ff85d514030-0&btsid=627f14df-6157-45b0-9595-d32cdf348a09&ws_ab_test=searchweb0_0,searchweb201602_10,searchweb201603_55

bys1123, 2019-11-29:
https://github.com/bys1123/upixels/blob/master/lc302_3c.cpp

My team wrote a draft version for PX4 1.9.2, still need time to port to 1.10, welcome to help us.

dagar, 2019-11-29:
If you open a pull request against current master I'll help get it into shape to merge.

hamishwillee, 2019-12-01:
Also try create some docs on usage for the user guide please once it is in: http://docs.px4.io/master/en/sensor/px4flow.html

bys1123, 2020-04-22:
@dagar PR number is #14723 , I need your help when you have time, thanks!

Last-ID, 2022-10-23:
Will this plan continue **in the future**?

dagar, 2022-10-23:
@Last-ID are you looking for support for this particular sensor or optical flow in general? In general optical flow support has continued to major in PX4, but often using the Pixart sensors directly or through CAN.

Last-ID, 2022-10-23:
@dagar Thank you very much for your reply :>
    This optical Pixel module should be able to obtain data to the **UORB** message pool through **serial port** connection, and with the external **TOF** module to obtain the height, it should be relatively easy to use.However, since I have been working on this project for a relatively short time, there are some difficulties in **transplanting**.
    As mentioned by the owner of this issue, LC302 is a more popular Pixel module, and seeing that the completeness of the porting in his project seems to be quite high(Although it is not clear why the merger was not successful), hopes to be merged into the project.