PX4/PX4-Autopilot #13635: Navio2 ESC calibration and PWM Outputs parameters missing

ClaudioCimarelli, 2019-11-29:
**Describe the bug**
 I am experiencing an incompatibility between Navio2 and PX4 in controlling the PWM output to the esc. 
- In particular, is not possible to make a successful ESC calibration since the device file ``/dev/pwm_output0`` does not exist ( there is a drive here `` /sys/class/pwm/pwmchip0 ``) and the result is that the motors do a strange sound when arming. 
- Moreover, 90% of the parameters in the PWM Output category do not exist for the Navio.

This issue is related to the closed #11743, that was solved by switching to APM...(I can confirm that it works without problem, but it is not a solution for me).

I am wondering how could I turn the driver under `` /sys/class/pwm/pwmchip0 `` into the expected device file and if anyone ever was able to fly using PX4 with a NAVIO2 board.

**Drone:**
- S500 frame with a Navio2 autopilot.
- 4 Blheli_s 32bits 30A ESC
- 4 T-motor MN3110 780KV

Thank you for your help in avance, hoping this issue will have better luck than #11743 :crossed_fingers: :smile:

dagar, 2019-11-29:
Unfortunately support for the Navio2 was never brought to the same level as the pixhawk flight controllers and infrastructure wasn't put in place to keep it working well.

dagar, 2019-11-29:
What does your bootup look like in PX4? Is `linux_pwm_out` starting successfully? What does `pwm info` show?

ClaudioCimarelli, 2019-11-29:
I receive an error in both cases:

[image]

I tried the module for esc calibration directly but of course not a chance:
[image]

I naively used this driver under sysfs the same way as the expected device file. I am totally new to drivers, so I am trying everything. Anyway somewhere in the code of PX4 there is already the key to dialogue using navio pwm drivers, but I do not know where to look yet.

ClaudioCimarelli, 2019-11-29:
In addition, there is a strange issue with the Navio2 rebooting after the disarming. I have not managed to localize the config which causes this because this does not happen always.

SalimTerryLi, 2019-11-30:
I created a post on px4discuss just now. [Here](https://discuss.px4.io/t/parameter-implements-about-pwm-output/14113)

ClaudioCimarelli, 2019-12-02:
Update: I have noticed that the module Linux Pwm Out actually starts fine. It was my error that it was already running and thus not starting again. Pwm module does not start anyway, but I think it is normal for navio.  

I think the problem is setting the correct period value for the pwm device and the duty cycle.  I have checked that with ardupilot the period is set to 2040816 ns by printing the file ``/sys/class/pwm/pwmchip0/pwm0/period `` while PX4 set it to 2500000 ns. This value should probably be read from the PWM_RATE param but it is not present like others in the same group.

tgadi001, 2020-07-16:
@ClaudioCimarelli were you able to find any work around for this problem? I have same setup as your with px4 running on a RPi with NAVIO2 shield, but my ESCs are not calibrated  so i can not fly my vehicle. 
I was thinking to install APM on a different SD card, plug it in to the RPi, connect to Mission Planner, perform ESC callibration on Mission Planner, then write down PWM min/max values for each channel and finally plug back in the SD card with px4 to the RPi and manually enter the min/max PWM values as the parameters on px4. Only problem is that i don't think changing `PWM_MAIN_MINx` and `PWM_MAIN_MAXx` parameters on px4 flight stack does anything. 

Another thing I was wondering was if there is any way to fix the current `esc_calib` function on px4 flight stack? maybe instead of trying to access the `/dev/pwm_output0` there is a way to make it access the right file, i just haven't been able to find any useful information cause i am really new to this.

Anyone's thoughts or comments would be really appreciated. Even if it doesn't give me the right solution at least i would know what not to waste my time on.
Thank you in advance!!

SalimTerryLi, 2020-07-16:
> @ClaudioCimarelli were you able to find any work around for this problem? I have same setup as your with px4 running on a RPi with NAVIO2 shield, but my ESCs are not calibrated so i can not fly my [...]

We need a new driver to replace [this](https://github.com/PX4/Firmware/blob/master/src/drivers/linux_pwm_out/linux_pwm_out.cpp), with new framework and new parameters implemented. That old driver doesn't worth any updating.