PX4/PX4-Autopilot #13661: Cannot increase IMU publish rate in Omnibus F4 SD

Razbotics, 2019-12-02:
There is no etc/ folder in mounted SD card so I tried making /etc/extras.txt and adding 
''' mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200 '''
but no luck, this seems to work on Pixhawk but not on Omnibus

bkueng, 2019-12-03:
What's the output of `dmesg`?

Razbotics, 2019-12-04:
> What's the output of `dmesg`?

Can you please elaborate which information you want using dmesg? Or you just want the whole batch of infomation outputting in my console?

```
[    0.000000] microcode: CPU0 microcode updated early to revision 0x2d, date = 2019-03-07
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.0-141-generic (buildd@lcy01-amd64-006) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) ) #167-Ubuntu SMP Wed Dec 5 10:40:15 UTC 2018 (Ubuntu 4.4.0-141.167-generic 4.4.162)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-141-generic.efi.signed root=UUID=9783446b-897c-4fc4-af73-800df48b9d21 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000009c2bdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000009c2be000-0x000000009c4bdfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000009c4be000-0x000000009cdbdfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009cdbe000-0x000000009cfbdfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009cfbe000-0x000000009cffdfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009cffe000-0x000000009cffefff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffa00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000025effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  SMBIOS=0x9c631000  ESRT=0x9c73d118  ACPI 2.0=0x9cffd014 
[    0.000000] esrt: Reserving ESRT space from 0x000000009c73d118 to 0x000000009c73d150.
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: HP HP Pavilion Notebook /8096, BIOS F.84 05/25/2017
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x25f000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-E7FFF write-protect
[    0.000000]   E8000-EFFFF write-combining
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 7C00000000 write-back
[    0.000000]   1 base 009CFFF000 mask 7FFFFFF000 uncachable
[    0.000000]   2 base 009D000000 mask 7FFF000000 uncachable
[    0.000000]   3 base 009E000000 mask 7FFE000000 uncachable
[    0.000000]   4 base 00A0000000 mask 7FE0000000 uncachable
[    0.000000]   5 base 00C0000000 mask 7FC0000000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] original variable MTRRs
[    0.000000] reg 0, base: 0GB, range: 16GB, type WB
[    0.000000] reg 1, base: 2572284KB, range: 4KB, type UC
[... 1016 lines cut ...]
[  122.521220] Bluetooth: RFCOMM socket layer initialized
[  122.521227] Bluetooth: RFCOMM ver 1.11
[  130.447604] logitech-djreceiver 0003:046D:C52B.0007: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input2
[  131.563360] input: Logitech Wireless Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0007/0003:046D:4055.0008/input/input18
[  131.564045] logitech-hidpp-device 0003:046D:4055.0008: input,hidraw1: USB HID v1.11 Mouse [Logitech Wireless Mouse] on usb-0000:00:14.0-2:1
[  750.241977] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  750.371029] usb 1-1: New USB device found, idVendor=26ac, idProduct=0016
[  750.371033] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  750.371035] usb 1-1: Product: PX4 OmnibusF4SD
[  750.371037] usb 1-1: Manufacturer: Vertile
[  750.371039] usb 1-1: SerialNumber: 0
[  750.371190] usb 1-1: ep 0x83 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[... cut ...]
[  755.474586] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  755.604089] usb 1-1: New USB device found, idVendor=26ac, idProduct=0001
[  755.604093] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  755.604096] usb 1-1: Product: PX4 OmnibusF4SD
[  755.604098] usb 1-1: Manufacturer: Vertile
[  755.604099] usb 1-1: SerialNumber: 0
[  755.604227] usb 1-1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[  755.604703] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```

bkueng, 2019-12-04:
Yes, but on the [mavlink console](https://docs.qgroundcontrol.com/en/analyze_view/mavlink_console.html), not your system

Razbotics, 2019-12-04:
> Yes, but on the [mavlink console](https://docs.qgroundcontrol.com/en/analyze_view/mavlink_console.html), not your system

Strangely, it shows no command found

bkueng, 2019-12-04:
Can you flash master?

Razbotics, 2019-12-04:
@bkueng Here is the output of dmesg after flashing master
```
nsh> dmesg
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: OMNIBUS_F4SD
FW git-hash: bd98a4eebbee750a4db007b03bcd1e40896cb5d5
FW version: 1.10.0 80 (17432704)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 398a59aaa4df2b32af2fdc1e3dd566f88d7eea38
Build datetime: Dec  4 2019 14:57:58
Build uri: localhost
Toolchain: GNU GCC, 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
PX4GUID: 000100000000203135484b4b500e003c0023
MCU: STM32F???, rev. 1
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
WARN  [rgbled] no RGB led on bus #2
ERROR [param] Parameter LIGHT_EN_BLINKM not found
Board sensors: /etc/init.d/rc.board_sensors
MPU6000 on SPI bus 1 at 1 (1000 KHz)
WARN  [hmc5883] no device on bus 2 (type: 2)
BMP280_SPI on SPI bus 3 at 1 (10000 KHz)
ERROR [param] Parameter TRIG_MODE not found
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Onboard, data rate: 46080 B/s on /dev/ttyS1 @ 921600B
ERROR [mavlink] offboard mission init failed (-1)
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter MNT_MODE_IN not found
ERROR [param] Parameter TEL_BST_EN not found
Board extras: /etc/init.d/rc.board_extras
Addons script: /fs/microsd/etc/extras.txt
WARN  [mavlink] mavlink for device /dev/ttyACM0 is not running
WARN  [mavlink] mavlink for device /dev/ttyACM0 is not running
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [mavlink] Starting mavlink shell
nsh> 
```

bkueng, 2019-12-05:
It should be fixed with https://github.com/PX4/Firmware/pull/13686. Can you test it?

Razbotics, 2019-12-05:
@bkueng I did couple of things,
I added `mavlink start -d /dev/ttyACM0 -b 921600 -m onboard` at top of extras.txt before 
`mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200` command.
Also, modified `mavlink_main.cpp` in  `src/modules/mavlink` by changing stream rates of HIGHRESS_IMU to 200HZ at Onboard section. Since, mavlink is now running at Onboard mode.  
After doing these things I could obtain 200HZ publish data in ROS.

closed 2019-12-06 by bkueng