PX4/PX4-Autopilot #13667: ST16 "manual control lost" after decoding initial stick positions

JoshG20, 2019-12-03:
I'm working on using the ST16 Controller with pixhawk. The receiver is the SR24, also used with the ST24, which is a protocol included with pixhawk. I use QGC and connect the pixhawk via USB, and then connect the SR24 receiver, which is already bound to the ST16 controller). Once the receiver is connected to the spkt/dsm port, all 12 channels appear with the correct stick positions in the channel monitor, but QGC immediately warns "Manual control lost" and the channels do not react to changes in stick position. This can be reproduced by unplugging and re-inserting the SR24 receiver again, but does not occur when starting pixhawk with the SR24 connected. I had hoped to figure this out myself but I do not have the coding knowledge to debug the ST24 protocol implementation or debug in QGC. With my limited coding knowledge my best guess is that the RC test is working correctly and it is decoding at least one packet, but there is a bug in the decoder itself. I can do any testing you need as long as you show me how.

julianoes, 2019-12-06:
Is this the same as #13265?

JoshG20, 2019-12-06:
This could very well be the same issue. I don't exactly know how GitHub works, is this fix incorporated in the most recent stable version, or just this person's branch?

julianoes, 2019-12-06:
This pull request might potentially fix it: https://github.com/PX4/Firmware/pull/13266

Can you flash this branch and try if it fixes your issue?

JoshG20, 2019-12-07:
I'm trying to do that, but as I said, I'm new to coding and github. I tried to compile the code but it's giving me the error:
```
$ make px4_fmu-v5_default
make: *** No rule to make target 'px4_fmu-v5_default'.  Stop.
```
I will keep trying but I would really appreciate if someone helped compile the code. I'm using a px4 board by the way.

julianoes, 2019-12-16:
@JoshG20 are you in the Firmware directory when you run this command?

JoshG20, 2019-12-16:
Yes, should I try switching to linux? I have a raspberry pi to give it a shot. Also, I realized I my board is  px4_fmu-v2_default, but it gave the same result.

julianoes, 2019-12-17:
I would be surprised if you can build on a Raspberry Pi! That would require the toolchain to work on the arm architecture, etc.

You should be able to use Windows or macOS.

JoshG20, 2019-12-17:
Ah didn't know raspberry pi wasn't supported. I still can't get the windows compiler to work, got any ideas?

julianoes, 2019-12-17:
> I still can't get the windows compiler to work, got any ideas?

Are you following this? https://dev.px4.io/master/en/setup/dev_env_windows_cygwin.html

JoshG20, 2019-12-17:
Yep. Also tried running the sim, same error message.

julianoes, 2019-12-17:
Have you done `cd Firmware` to change directory to firmware?

JoshG20, 2019-12-17:
Yes, I'm in the correct directory.

julianoes, 2019-12-19:
Can you add a screenshot or copy paste the commands you're doing? I can't help otherwise.

julianoes, 2020-03-20:
No follow up, closing.

JoshG20, 2020-05-23:
Hello, I know this was closed due to inactivity, but I finally managed to implement all of the changes made by Ultrasystem onto the v1.10.0 release and compile correctly. The firmware runs fine in jmavsim and on the drone but the receiver issue still persists on both of my identical receivers. The ST16 shows a correct bind to the sr24 receiver and the light is solid, but PX4 only decodes the initial positions. Unlike Ultrasystem, the issue is not repeating over time, it says manual control lost but never reconnects until you restart the entire system. Any ideas of what could cause this?

JoshG20, 2020-05-23:
Interestingly, the stick inputs are correctly decided in HITL mode and I was able to calibrate them. However, it does sometimes decide to throw the manual control lost error after a few minutes. Once it throws the error the only way to fix it is to restart the entire drone.

julianoes, 2020-05-25:
It's hard to say what causes the lockup. If you don't have ST16 connected, does the lockup not happen? And have you checked the ST16 (st24) parsing for any infinite loops that might happen or stuff like that?

The only thing at this point to suggest is to connect a serial console to debug it or even JTAG to debug it with gdb. This all depends on what hardware you have available.

JoshG20, 2020-06-13:
I believe I have found a hint of the issue. In HITL mode, the manual control lost seems to be triggered by the obstacle avoid switch being thrown. Control is resumed after approximately 500ms.  I had attempted to write my own decoder in the past and noticed that whenever the switch was triggered it would affect multiple serial packets, so I wouldn't be surprised if this is causing a problem. However, when I leave HITL mode it just says manual control lost no matter what position that stick is in and never reconnects.

JoshG20, 2020-06-13:
I also checked that a normal 6ch PPM transmitter has no issues, and it seems to work perfectly, went through calibration and it seems stable (running normally on aircraft, not HITL)

JoshG20, 2020-06-16:
Some more interesting info I noticed: the decoding works perfectly for just under one second even in normal running(non-HITL). Stick movements are being displayed accurately up until that point, where is says "manual control lost". Now I'm looking for reasons why it may timeout in the code and never recover until you powercycle the autopilot. I'm currently running v1.10.1 code with ultrasystem's fixes.

JoshG20, 2020-06-17:
I've also gotten it to loose manual control in HITL mode by doing the sticks down-out arm command, which makes no sense to me.

JoshG20, 2020-06-17:
Also interesting, by unplugging it is the short period of time between it working and the error, I managed to get it to reconnect twice before fully locking out. , as shown by this log. [](https://logs.px4.io/plot_app?log=4b3d421c-098c-47c8-80ae-93326d100dfa)

LorenzMeier, 2021-01-01:
Please comment if this is still an issue for you.

closed 2021-01-01 by LorenzMeier