PX4/PX4-Autopilot #13669: LPE_FUSION does not show up

netgittemp, 2019-12-03:
Hi, I am using PX MINI 4, and I am integrating PX4 with companion computer and PX4FLOW.
To do that, I am changing parameters using QGC with usb connection.
However, I cannot find `LPE_FUSION` parameter.

How can I set `LPE_FUSION` parameter?

Thank you.

TSC21, 2019-12-04:
@netgittemp we have discontinued LPE. Please use EKF2 instead and set the correct fusion flag for it - https://dev.px4.io/master/en/advanced/parameter_reference.html#EKF2_AID_MASK. The OF params that you can tune have the `EKF2_OF_` suffix, but I advise you to use the default ones for now.
To use the PX4Flow, make sure that for your your board you have `SENS_EN_PX4FLOW` set to `1` and that `SENS_FLOW_ROT` matches the correct sensor orientation of your board with respect to the aircraft frame.

netgittemp, 2019-12-04:
Hi @TSC21 ,
Thank you for your kind reply.
It is very helpful. Thank you so much.

I have one more issue, and it is kind of bug report.

In [Return Mode](https://docs.px4.io/v1.9.0/en/flight_modes/return.html) page, it is described that the quadrotor goes up to [RTL_RETURN_ALT](https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html#RTL_RETURN_ALT) and move to the home position, then the quadrotor descend to the [RTL_DESCEND_ALT](https://docs.px4.io/v1.9.0/en/flight_modes/return.html#RTL_DESCEND_ALT). When the quadrotor reach to [RTL_DESCEND_ALT](https://docs.px4.io/v1.9.0/en/flight_modes/return.html#RTL_DESCEND_ALT), it will hold the position for [RTL_LAND_DELAY](https://docs.px4.io/v1.9.0/en/flight_modes/return.html#RTL_LAND_DELAY) before landing.

For your information, I set the RTL_* parameters as shown in the following figure;

[image]

Could you check it for me?
If you would like to make this as bug report, Then I will make another post for this issue.

Thank you.

closed 2019-12-13 by netgittemp