PX4/PX4-Autopilot #13671: Flight Speed control doesn't always work

mwiatt, 2019-12-03:
**Describe the bug**
When a "Flight Speed" is attached to a waypoint in QGC, the speed is not always used. This behavior is intermittent but happens more often than not.

[image]

**To Reproduce**
This behavior is intermittent.

Steps to reproduce the behavior:
- Create mission with a "Flight speed" attached to a waypoint
- Upload to flight controller
- Take off and fly in mission mode.

**Expected behavior**
New speed is used.

**Log Files and Screenshots**
- Log: https://logs.px4.io/plot_app?log=36040896-dbc3-48d4-baab-ec18c0ed29bd
- Plan: [no-speed-control.plan.txt](https://github.com/PX4/Firmware/files/3918981/no-speed-control.plan.txt)

**Drone (please complete the following information):**
- Tarot X6 (with Tarot retracts)
- Pixhawk Cube 2.1
- Here2 GNSS
- PX4 1.10.0 rc2
- QGC daily 12/2/2019 Win10
- Gremsy S1 Gimbal connected to Telem2 (Mavlink)

**Additional context**
This feels **very** similar to this bug https://github.com/PX4/Firmware/issues/13474 In fact, I had an ROI waypoint in the plan and speed control and both were sometimes ignored. If one worked, the other seemed to usually work. If the ROI failed, the speed control would typically fail also. It seems like the messages from the piece of the firmware that processes the waypoints are often ignored.

mwiatt, 2019-12-03:
Below is a log from a flight a few minutes earlier than the one above.

https://logs.px4.io/plot_app?log=a8662dfc-e6f1-453a-9c9d-8ca576b93340

This uses the same exact plan file. You can see that the 3 m/s speed is obeyed this time (half way through the log file, look at X velocity).

[image]

From the previous flight where the 3 m/s was ignored:

[image]

bresch, 2020-03-05:
@julianoes IIRC you are testing different velocities in the MAVSDK test. Have you spotted this issue as well?

julianoes, 2020-03-05:
@bresch yes there are tests for that, and I think they are passing.

https://github.com/mavlink/MAVSDK/blob/4d8d4b3836a2bc46c01b89bfd523306ef7c851a7/src/integration_tests/mission_change_speed.cpp#L62-L64

dayjaby, 2020-05-18:
Probably the same cause as https://github.com/PX4/Firmware/issues/13474, where the navigator_main is not able to handle too many vehicle_commands at once on real hardware.

In the above logfile, the command is surrounded by other commands:

```
... lots of other gimbal commands
346654899,0.0,0.0,2.0,0.0,0.0,0.0,0.0,204,1,154,0,0,0,0
346654909,0.0,0.0,-0.0,0.0,-0.0,0.0,2.0,205,1,154,0,0,0,0
346661539,34.23981475830078,-119.05514526367188,0.0,0.0,0.0,0.0,69.73837,195,1,1,1,1,0,0
346678523,0.0,0.0,2.0,0.0,0.0,0.0,0.0,204,1,154,0,0,0,0
346678927,0.0,0.0,-0.0,0.0,-0.0,0.0,2.0,205,1,154,0,0,0,0
... lots of other gimbal commands
```
so the navigator_main.cpp has only a very short timeframe to read this particular command, making it very likely that the command got overwritten in the meantime.