PX4/PX4-Autopilot #13675: Yaw during takeoff

julianoes, 2019-12-04:
**Describe the bug**
Sometimes I observe a multicopter doing a strange yaw back and forth right during takeoff.

**To Reproduce**
Do many takeoff in SITL and every now and then it's visible.

**Expected behavior**
It should not yaw during takeoff, period.

**Log Files and Screenshots**
[image]
[image]

https://logs.px4.io/plot_app?log=907be591-f04b-4187-8e5a-153435b0bfe1
https://logs.px4.io/plot_app?log=bb0b86a5-a23c-4acd-a88a-fda4a8b6b78c

**Drone (please complete the following information):**
- The logs are from a VTOL takeoff without weathervane activated, however, I've also seen the behaviour with iris.

mwiatt, 2019-12-04:
On takeoff, I have seen a yaw of around 5 - 10 degrees after the copter reaches around 2m in alt. I don't remember similar behavior with 1.9.2. Not sure if this is the same bug.

See this log, around 5:28.65:

https://logs.px4.io/plot_app?log=36040896-dbc3-48d4-baab-ec18c0ed29bd

[image]

MaEtUgR, 2019-12-04:
I recently also saw a strange twitch in SITL on master when doing takeoff. Just a wild guess because of what changed recently: Might be related to https://github.com/PX4/Firmware/pull/13430. Someone up to verify until I find the time to do tests myself?

mwiatt, 2019-12-04:
I was looking back at some of my old flight logs and this is an interesting one from 1.9.2. You can see that the estimated yaw a step change on takeoff but the flight controller does not react to it.

https://logs.px4.io/plot_app?log=c57c04e0-a6bd-4f04-9fe2-eb986cbcc1b4

[image]

MaEtUgR, 2019-12-04:
There can be yaw resets from the estimator in case it realizes after takeoff that the global yaw heading estimate was not accurate e.g. because of magnetic interference. That could also be something realted to the issue here.

> but the flight controller does not react to it

What would you expect to see as a reaction in terms of this plot? Do you expect the setpoint to switch as well since you're flying in manual and basically don't care about the absolute global heading accuracy and want your vehicle to not correct for it?

mwiatt, 2019-12-04:
> Do you expect the setpoint to switch as well since you're flying in manual
I was flying in Mission mode at the time of the step change in yaw estimate. I wouldn't expect the yaw setpoint to change. I was just pointing out that 1.9.2 and 1.10.0 react differently to the step change. Sorry I wasn't clear.

MaEtUgR, 2019-12-06:
In the logs posted by @julianoes it's really strange that the yaw setpoint switches to zero even though the task produces no jumps. This means it has to be some logic of the position controller itself which should basically only forward yaw setpoints.
[image]

julianoes, 2020-01-23:
Fixed in #13965.

closed 2020-01-23 by julianoes