PX4/PX4-Autopilot #13682: VTOL transitons to MC immediately after RTL

sfuhrer, 2019-12-05:
**Describe the bug**
VTOL in mission mode: when RTL is triggered after the vehicle has already reached the 2nd to last waypoint (thus having as next wp the landing waypoint), it immediately transitions to multicopter instead of proceeding flying in fixed-wing until close to landing (as it does when RTL was triggered earlier in the mission, with not landing currently as active wp).

**To Reproduce**
Plan mission in SITL with landing at end, and long distance between landing and prior wp. Trigger RTL when 2nd to last wp is already reached.

**Expected behavior**
VTOL should proceed flying in FW until close to home/land.

**Additional context**
Long flights in MC are bad for battery consumption, which is especially problematic during RTL situations where low battery s could have been the cause.

Note: this problem shouldn't be present when the mission is planned in the recommended manner for VTOL with a DO_LAND_START marker, a loiter to altitude and then a landing wp. Maybe we should thus now try to enable to set this landing procedure in a simple manner for the pilot (similar to how it is done for FW systems).

FYI @RomanBapst

julianoes, 2019-12-05:
This only happens with `RTL_TYPE = 1`. From my SITL testing it looks like the normal RTL procedure is ok.

sfuhrer, 2019-12-18:
Yes only happens with RTL_TYPE = 1. Unfortunately this IS the normal RTL procedure for VTOLs ;)

closed 2021-01-18 by sfuhrer