PX4/PX4-Autopilot #13685: Logger complains about too many subscriptions

julianoes, 2019-12-05:
On SITL startup with `standard_vtol` I see the following logger warnings:

```
INFO  [logger] logger started (mode=all)
WARN  [logger] Too many subscriptions, failed to add: sensor_selection 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry_aligned 0
WARN  [logger] Too many subscriptions, failed to add: cellular_status 0
WARN  [logger] Too many subscriptions, failed to add: collision_constraints 0
WARN  [logger] Too many subscriptions, failed to add: obstacle_distance_fused 0
WARN  [logger] Too many subscriptions, failed to add: onboard_computer_status 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_mocap_odometry 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry 0
```

FYI @bkueng

bkueng, 2020-03-04:
This was fixed, the array size is now dynamic.

closed 2020-03-04 by bkueng

AlexWUrobot, 2023-01-24:
I use the latest PX4-autopilot 
My platform is ubuntu 18.04 and python 3.6
I have the same issue when run 
roslaunch px4 mavros_posix_sitl.launch 

[image]

marcoCella, 2023-02-14:
> I use the latest PX4-autopilot My platform is ubuntu 18.04 and python 3.6 I have the same issue when run roslaunch px4 mavros_posix_sitl.launch
> 
> [image]

Hi! I’m having the same issue, did you find a solution?

blax185, 2023-03-20:
I have the same issue when run :
roslaunch px4 mavros_posix_sitl.launch
I use Ubuntu 20.04 and latest version of PX4-Autopilot

[image]

biagio-trimarchi, 2023-04-04:
I have the same issue.

I am on Ubuntu 20.04.6 LTS.

When I run "make px4_sitl_default gazebo-classic" I get the following log:

[0/4] Performing build step for 'sitl_gazebo-classic'
ninja: no work to do.
[3/4] cd /home/casy/PX4/PX4-Autopilot/build/px4_...me/casy/PX4/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/casy/PX4/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
model: iris
world: none
src_path: /home/casy/PX4/PX4-Autopilot
build_path: /home/casy/PX4/PX4-Autopilot/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/casy/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic
GAZEBO_MODEL_PATH :/home/casy/PX4/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models
LD_LIBRARY_PATH /opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu:/home/casy/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic
empty world, setting empty.world as default
Using: /home/casy/PX4/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
SITL COMMAND: "/home/casy/PX4/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/casy/PX4/PX4-Autopilot/build/px4_sitl_default"/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10015
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 316 bytes, decoded 316 bytes (INT32:13, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number: 
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 3096000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.213.254
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 0
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 1
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 2
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 3
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 4
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_ev_yaw 5
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_gnss_hgt 0
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_gnss_hgt 1
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_gnss_hgt 2
WARN  [logger] Too many subscriptions, failed to add: estimator_aid_src_gnss_hgt 3
[... 50 lines cut ...]
WARN  [logger] Too many subscriptions, failed to add: ekf2_timestamps 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry 0
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 2
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 3
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 4
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 5
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 6
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 7
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 8
WARN  [logger] Too many subscriptions, failed to add: distance_sensor 9
WARN  [logger] Too many subscriptions, failed to add: collision_constraints 0
WARN  [logger] Too many subscriptions, failed to add: obstacle_distance_fused 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_mocap_odometry 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_visual_odometry 0
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2023-04-04/12_37_49.ulg	
INFO  [logger] Opened full log file: ./log/2023-04-04/12_37_49.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
ERROR [flight_mode_manager] Failsafe task activated
INFO  [commander] Ready for takeoff!