PX4/PX4-Autopilot #13688: Drop when switching from MANUAL to OFFBOARD on the flight

negvet, 2019-12-06:
**Describe the bug**
A drone suddenly drops down 0.5-1 meters when switching to OFFBOARD on the flight.

The problem is that for some reason when I switch to OFFBOARD for the very small first period of time the `POSITION_TARGET_LOCAL_NED.vz` is positive (so the drone goes down). It is also observable in `/mavros/setpoint_raw/target_local`. The very first message after switching to offboard:

> header: 
>   seq: 196
>   stamp: 
>     secs: 1455209561
>     nsecs: 290099086
>   frame_id: ''
> coordinate_frame: 1
> type_mask: 0
> position: 
>   x: -0.06703799963
>   y: 0.402279406786
>   z: 1.61291372776
> velocity: 
>   x: 1.2246468174e-17
>   y: -2.46519036555e-33
>   z: -0.50000000149
> acceleration_or_force: 
>   x: -0.0379852429032
>   y: 0.0017760390183
>   z: 4.83928060532
> yaw: 1.6123470068
> yaw_rate: 0.0

I do not know how to fix it.

When I set to OFFBOARD on the ground - everything works just fine, I am able to take off, fly around and land.

PX4 Firmware: 1.8.2.
Cube flight controller

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize EKF2 on the ground (vision is available). Publish vision pose and orient to `/mavros/vision_pose/pose_cov`. 
2. Move the drone nearby on the ground, where vision is not available.
3. Takeoff in manual mode with RC.
4. Fly to the area where vision is available (1.5-2 meters height). 
5. If vision is available -> grab position and orient from `/mavros/local_position/pose` and publish several setpoints to `/mavros/setpoint_position/local`
6. Switch to offboard.
7. The drone suddenly goes down 0.5-0.7 m but then stops, goes back up and follows the setpoints.

**Expected behavior**
I expect the drone to hover at the position where and when the OFFBOARD is enabled.

**Log Files and Screenshots**
[image]

julianoes, 2019-12-16:
@negvet have you tried if this issue persists in 1.9.2 or 1.10?

negvet, 2019-12-16:
Sorry but I was not able to set up the latest firmware for my application
due to separate issues.

For now, I minimized the effect if this issue by decreasing the maximum
vertical acceleration and velocity parameters.

julianoes, 2019-12-17:
@negvet so with what version was that?

negvet, 2019-12-18:
PX4 Firmware: 1.8.2.

julianoes, 2019-12-19:
@negvet ok it's likely that this has been fixed in 1.9 or 1.10. I would suggest to try with the newer releases.

julianoes, 2020-03-20:
No follow up, closing.

closed 2020-03-20 by julianoes