PX4/PX4-Autopilot #13697: Bad system behavior when airspeed data disapears

Antiheavy, 2019-12-06:
Here is some bad system behavior we've observed in legacy firmware (between v1.8-v1.9) related to airspeed data disappearing, for example if the I2C bus flakes out or if power is lost to the airspeed sensor.

I submit this for review against the new airspeed selector work @sfuhrer and others are contributing in hopes the new airspeed features will robustly handle these difference cases.

If airspeed data suddenly disappears there are several bad behaviors:

1. The system seems to keep sending the last known airspeed reading down the MAVlink pipe to the ground station (might need to be verified).  It should instead stop sending airspeed data if no new data is available.

2. There are no warnings to the user that airspeed data has gone missing. There should be some warning inidcation if airspeed data is expected but disappeared.  There should also be an indication if it returns.  (note: this type of data disappeared, data returned indications should be extended to all sensor types).

3. On a failed takeoff, if the autopilot stops receiving airspeed data, the LandDetector may think the vehicle is still flying because the last reported airspeed seems to be piping into the system. The motor may still run at full throttle! Ideally airpseed would stop updating and "zero" or "null" would be reported and the system would understand that it should not be using airspeed as part of the LandDetector.

4. If airspeed disappears during flight there is no indication that it disappeared. TECS will silently switch to non-airspeed mode and not tell the user and by-passes the airspeed failsafe functionality. There should be warning indications and the airspeed failsafe functionality should be followed.

Here are some flight logs showing the last example of airspeed data disappearing during the middle of a flight:
https://review.px4.io/plot_app?log=d378dac5-1061-444d-ba8d-ee911c6315cf
https://review.px4.io/plot_app?log=35be23c2-6bdf-4caa-9075-f9805b6d2035
https://review.px4.io/plot_app?log=5ffc7018-9793-4b88-89a3-8066988b1e54

FYI @sfuhrer

dagar, 2019-12-09:
Some of the notification behavior should be improved with the airspeed selector, but these other problems still exist.

Basically, we need to a do a thorough pass to check every single use of airspeed in the system and make sure the timeout case is explicitly handled.

For example, airspeed used in the land detector. https://github.com/PX4/Firmware/blob/904ab16558272939f15a4ce90255783a761dc1f9/src/modules/land_detector/FixedwingLandDetector.cpp#L86

dagar, 2019-12-09:
Here's a quick fix for 1) and 3). https://github.com/PX4/Firmware/pull/13706

sfuhrer, 2019-12-09:
Thanks for raising these issues @Antiheavy . I'll quickly address your 4 points in respect to what the current implementation does:

1. The new airspeed validator detects missing data, and sets the airspeed_validated to NAN. QGC currently displays the unvalidated airspeed data, this should be changed to the airspeed_validated/true_airspeed.

2. The user is informed if the airspeed has become invalid, and what action is done because of that. The message is though not very intuitive and informative, it says to which sensor it switches (where 1 is the primary airspeed sensor, 0 a groundspeed - windspeed airspeed approximation, and -1 means airspeed disabled). The airspeed validation also has the ability to recover after being set invalid, in which case it then says "Switched from -1 to 1". We could ofc make this more user-friendly, but for development these messages are already quite useful.
[image]

3. Currently the land detector doesn't consume the validated airspeed - we should change that! For the time being: let's use @dagar 's fix https://github.com/PX4/Firmware/pull/13706. 

4. See my answer for 2. The validator detects invalid airspeed data --> declares airspeed invalid, publishes it to NAN --> TECS switches to non-airspeed mode. 

One of the next steps I had in my mind was to get rid of CBRK_AIRSPD_CHK and FW_ARSP_MODE. The functionality they offer should all be handled inside the airspeed selector module (thus ASPD_XX params). Eg there should be just one parameter to be set when one plans to fly without airspeed sensor (whiich is ASPD_PRIMARY). 

Let's discuss any other improvements/ other features necessary here.

1Gump, 2019-12-11:
> One of the next steps I had in my mind was to get rid of CBRK_AIRSPD_CHK and FW_ARSP_MODE. The functionality they offer should all be handled inside the airspeed selector module (thus ASPD_XX [...]

Since i employ the airspeed sensor on the wing, and the wing is usually removed when the craft is not in flight or at the field, it is useful to temporarily disable the AS via a CBRK  when doing testing/setup/tweaking in house.

sfuhrer, 2019-12-15:
> Since i employ the airspeed sensor on the wing, and the wing is usually removed when the craft is not in flight or at the field, it is useful to temporarily disable the AS via a CBRK when doing [...]

The idea is to have the same ability in the future by changing temporarily ASPD_PRIMARY to "No airspeed mode". I haven't fully thought it through, but this is certainly a use case that I'll cover when I start implementing the changes required to (potentially) get rid of CBRK_AIRSPD_CHK and FW_ARSP_MODE.

1Gump, 2019-12-16:
Sounds like you're putting the time in to think it through, great. much appreciated.
.
For consideration as you proceed, and to clarify my point, it is useful (to me) to have an 'enable/disable' only menu for ALL my sensors. That way i dont have to know where the correct enable/disable switch is or worry about missing it, in the sea of parameters, when it's time to fly (or test). its easy to look at a single CBRK (or whatever it may be called) menu to verify on/off switches. Would be ok if there was a menu that 'referenced' the switches, and had edit capability.

sfuhrer, 2019-12-17:
@1Gump what other sensors do you then have to disable for bench testing?

1Gump, 2019-12-17:
@sfuhrer ...that's a good point. mainly the airspeed sensor. ive had to play with a couple other cbkrs before because of gps/compass problems and power setup. I haven't been doing anything for a couple/few months now because of building. will be employing gas engines soon instead of electric - not sure if the enginefail cbkr will be of any use.
.
anyways, the AS is, by far, the sensor I disable/enable the most here. the others not so much - only in weird situations. I wish they had a lidar cbkr here.

julianoes, 2021-02-09:
@sfuhrer is this still valid?

sfuhrer, 2021-02-09:
> @sfuhrer is this still valid?

I would say no, if somebody disagrees please reopen.

closed 2021-02-09 by sfuhrer