PX4/PX4-Autopilot #13702: frequent "Manual control lost" message when joystick is used in QGC

dimon777, 2019-12-09:
I am seeing lot of these warnings in QGC, when i have joystick connected (Screenshot attached). When I disconnect joystick and restart QGC the message never shows up (or shows up rarely). Also, when joystick is connected and "JoystickValuesLog" is enabled, the console output is continuously bombarded with joystick log messages. So, I suspect the issue might be that firmware is overloading mavlink when joystick is enabled. Thanks.

kpetrykin, 2019-12-10:
QGC by default sends joystick's values at 25 Hz when they are being changed and decreses the rate to 5 Hz when they aren't.
Try to increaze COM_RC_LOSS_T parameter because if it is less than 0.2 you will receive "Manual control lost" message and fail safe will be triggered.

dagar, 2019-12-10:
It would be nice to dig into the fundamental problem here and understand where the delay is coming in, but otherwise you can also increase `COM_RC_LOSS_T` slightly.

julianoes, 2020-03-26:
Closing because there was not follow up.

closed 2020-03-26 by julianoes

cwc2023, 2023-07-14:
cool