PX4/PX4-Autopilot #13703: NXP FMUK66- Telemetry 1 Port not working correctly.

LowOrbitIonCannon, 2019-12-09:
**Describe the bug**
Telemetry 1 Port not working Correctly.
While testing UWB functionality I encountered an issue with the Telem 1 Port.
The FMU sends the correct Data to the UWB, but overflows the buffer after a few seconds.
I don't get this issue on the Telem 2 Port.

**To Reproduce**
Telem 1 Port connected to the UWB device. (with RTS / CTS pulled to Ground )
Telem 2 Port connected to the UWB device. (with RTS / CTS pulled to Ground )

**Expected behavior**
On Terminal 2 I get the expected behavior of both the FMU and the UWB Device:
The correct command is send to the UWB device and data is send back Continuously.

I suspect that the FMU does not set CTS/RTS correctly and just sends control messages to the UWB Device until the Buffer Overflows.

**Log Files and Screenshots**
I get Buffer Overun Errors on the RX Buffer of the UWB device:
[image]

**Additional context**
@dk7xe suspects that CTS/RTS is not working properly  on Telemetry 1.
A module without CTS/RTS was working with this port eventhough they where not Pulled to Ground.

@ItsTimmy his code was used to Test UWB: [LINK](https://github.com/PX4/Firmware/pull/12956)
@davids5 could you have a look please?

ItsTimmy, 2019-12-11:
Would it be possible to use a logic analyzer / oscilloscope to verify for sure whether CTS/RTS are indeed the issue?

davids5, 2019-12-11:
@LowOrbitIonCannon 
From your post it is the Receive side of the UWB that is overflowing. 
[K66]-------TX----->[UWB]
[K66]CTS<------RTS[UWB]

I had a look at the SW and at the schematic: There are no HS lines on Telem2.  Telem1 uses the chip level HW HS.  This is not SW controlled, it is UART controlled per the data sheet:

> Enables clear-to-send operation. The transmitter checks the state of CTS each time it is ready to send
> a character. If CTS is asserted, the character is sent. If CTS is deasserted, the signal TXD remains in
> the mark state and transmission is delayed until CTS is asserted. Changes in CTS as a character is
> being sent do not affect its transmission

Please add [serial_test](https://github.com/PX4/Firmware/pull/13662)  to the the k66's default.cmake and test as described.  You can add -l 250 on each side:

>-l, --rx-delay    Delay between reading data (ms) (can be used to test flow control)

LowOrbitIonCannon, 2019-12-13:
I did that.

before i tested that i confirmed that i have a serial connection over this port to the Drone with QGC.
Everything worked fine in QGC

For testing i disabled the Mavlink instance on Telem1
and added serial_test to the FMU Build.
these are the messages i get when i start serial test on both ends:
[image]

On my Linux machine i get the same errors.
Both where configured to use 115200 Baudrate and i delay of 250.

dk7xe, 2020-11-16:
@davids5  any news on that one?

dk7xe, 2020-11-16:
@igalloway for you info