PX4/PX4-Autopilot #13717: Compass caliberation is required after every reset

Vishranthee, 2019-12-10:
After every reboot of PX4 via QGC, compass calibration is show as red though the Quad was not even moved during the process. IS this how calibration is supposed to work or am i missing something here ?

dagar, 2019-12-11:
That's definitely not normal. Parameters are saved to non-volatile storage.

Which board are you using? How are you rebooting? Are you losing any other parameters after reboot?

Can you look at the relevant calibration parameters before and after reboot? `CAL_MAG_*`

Vishranthee, 2019-12-13:
I'm using Pixhawk 4 and I'm rebooting via Parameters-> Tools-> Reboot vehicle in QGroundControl.

Compass caliberation is the only parameter that gets reset everytime I reboot it.  CAL_MAG_* changes from 3 to 0 after reboot.

Vishranthee, 2019-12-15:
@dagar Can you please have a look at this ?

closed 2020-01-08 by Vishranthee

LaurensKools, 2020-10-22:
@Vishranthee @dagar Currently I have the same issue. I'm using a Cube Orange with PX4 1.11.0, Here 2 GPS and Qgroundcontrol V4.0. Every time the system reboots (either via software or unplug the power), the compass calibration resets and the heading is inconsistent (pre-flight fail: Compasses inconsistent by x degrees). When recalibrating all sensors (gyro, accel, level horizon and compass) and not click "reboot" it does work fine. However, doing this every time after a reboot shouldn't be necessary so is there a solution to this problem?

dagar, 2020-10-22:
> it does work fine. However, doing this every time after a reboot shouldn't be necessary so is there a solution to this problem?

@LaurensKools are you losing all parameters or only compass calibration?

LaurensKools, 2020-10-22:
Thanks for your reply @dagar 
Only the compass calibration parameters reset to 0. However, I have to redo all calibrations to obtain a consistent heading estimate

yamgeva, 2021-03-03:
I'm having the same problems with 1.11.3. Any updates?