PX4/PX4-Autopilot #13719: Improving Safety Documentation  (All Failsafe Scenarios)

bozkurthan, 2019-12-10:
Hello,

I'm currently working on safety of PX4. And there are some gaps between documentation and PX4 stack. Especially there is only one mention about what is the scenario when 2 failsafe actions come up.

> **If a failsafe occurs while the vehicle is responding to another failsafe (e.g. Low battery while in Return mode due to RC Loss), the specified failsafe action for the second trigger is ignored. Instead the action is determined by separate system level and vehicle specific code. This might result in the vehicle being changed to a manual mode so the user can directly manage recovery.**

So I tried all scenario between failsafes' actions on the SiTL. Firstly I simulated 1 failsafe and waited to PX4 perform action of failsafe. Then I simulated another failsafe and I've checked to see if it's override first failsafe action. **Table 1** shows this overriding cases and some notes about all failsafes. Table shows that, for example, when "RC Loss" occurs and executes RC failsafe action, if "Low Battery" occurs then PX4 stops "RC Loss" action and executes "Low Battery" failsafe action.

**Table 1:** 2 failsafe status comparison when they occur together. 

[image]

NOTE 5: Issue number #4876

Tested on:
```
HW arch: PX4_SITL
FW git-hash: b8b7527d052a19dd31a51ded7cabd6d821ef67a5
FW version: 1.10.0 80 (17432704)
FW git-branch: master
OS: Linux
OS version: Release 5.0.0 (83886335)
Build datetime: Nov  6 2019 14:49:56
Build uri: localhost
Toolchain: GNU GCC, 7.4.0
PX4GUID: 1006554d49534954414c44494e4f3030303
```

If there is any specific documentation or issue please let me know.

Thanks!

Antiheavy, 2019-12-10:
This is fantastic!  

There are additional failsafes to consider:
- Airspeed loss (see @sfuhrer for the latest on this failsafe)
- Differentiate between Low, Critical, and Emergency battery levels
- probably something i'm forgetting...

A version of this table that could eventually make it into documentation should focus on the resulting action by the system.  Terminology of what "can" and "can't" override is useful here in the technical discussion, but would be confusing and unclear in documentation.

hamishwillee, 2019-12-10:
Yes, this is cool. 

The only other failsafe I know about is [Traffic avoidance](http://docs.px4.io/master/en/config/safety.html#traffic-avoidance-failsafe).

> A version of this table that could eventually make it into documentation should focus on the resulting action by the system. Terminology of what "can" and "can't" override is useful here in the technical discussion, but would be confusing and unclear in documentation.

I agree with this statement. Also agree that we need to break down the battery levels, because I doubt low battery warning should override data link loss. 

[image]

While "spelling it out" as you have done makes the relationship very clear, it also makes the diagram less succinct. Perhaps continue to use colour coding but just list the _winner_. E.g. above  **Critical battery failsafe**, **Data Link Loss**

hamishwillee, 2019-12-10:
PS I think we'd add this to a section at the end of http://docs.px4.io/master/en/config/safety.html

bozkurthan, 2019-12-11:
@Antiheavy @hamishwillee thanks for helping.
Unfortunately, I don't have any experience with FixedWings and Multi vehicle in neither simulation or neither field. So, I'm not be able to test these 2 failsafes. If someone could be able, it was wonderful. But, I added 3 battery levels. Battery level 1 is only sending warning message. So, all failsafes action override it. 
**Furthermore, I appreciated if someone can test in simulation and validate me.**

Also, I changed table to color format which make more sense to us and users, which is suggested by @hamishwillee. I deleted NOTE 4-5 but NOTE 1-2-3 should not be deleted because of special case.
Maybe 4-5 can be directly added to documentation related section.

NOTE 1-2-3:

NOTE 1: | There is a parameter that named   "COM_POSTCTL_NAVL" to catch RC will be avaible or not. But it's   first action for "Position Loss". If parameter value is   "Assume use of RC.." and when user in manual modes (Manual,   Stabilize, Acro, Ratitude,  Altitude or   Position)  than RC Loss failsafe action   triggered, vehicle failsafe action will be land. If vehicle mode wass set to   auto mode (Offboard, Hold, Mission or RTL), Regardless of parameter, vehicle   will be land.
-- | --
NOTE 2: | There are parameters those  named "COM_OBL_ACT" and   "COM_OBL_RC_ACT" to take action by vehicle mode.   "COM_OBL_RC_ACT" failsafe action changes to "COM_OBL_ACT"   when RC loss failsafe is triggered.
NOTE 3: | If "Geofence Violation" action is   selected "Hold Mode", then "Data Link Loss" failsafe   action can override, but if "Geofence Violation"action is selected   "RTL Mode" then "Data Link Loss" failsafe action can't override

NOTE 4-5

NOTE 4: | There is parameter that named   "NAV_DLL_N" to count number of allowed Datalink timeouts to take   action which is set for "Data Link Loss", but it is not mentioned   in document (v1.10 - stable)
-- | --
NOTE 5: | "Data Link Loss" is not working   following modes; "POSCTL / ALTCTL / MANUAL / STABILIZE".

LATEST TABLE:
[image]

Maybe colors could be assigned wrongly when was converting. :)

hamishwillee, 2019-12-11:
Thanks. Can you share the source doc with me? What I think I will do is publish something similar in docs. For now without the airspeed and traffic avoidance failsafes.

@Antiheavy Is there a way to simulate airspeed loss?

bozkurthan, 2019-12-12:
Sure,
Here xlsx table:
[GITHUB TABLE.xlsx](https://github.com/PX4/Firmware/files/3953936/GITHUB.TABLE.xlsx)

Antiheavy, 2019-12-12:
> @Antiheavy Is there a way to simulate airspeed loss?

Last I tried there wasn't an easy way to simulate airspeed loss, but that was a year ago.  I wonder if @sfuhrer knows of a way that it can be done?

hamishwillee, 2019-12-16:
Thank you. I'm pretty busy this week, but will try get this in. If not, when I'm back in Jan.

sfuhrer, 2019-12-17:
> > @Antiheavy Is there a way to simulate airspeed loss?
> 
> Last I tried there wasn't an easy way to simulate airspeed loss, but that was a year ago. I wonder if @sfuhrer knows of a way that it can be done?

Yes there is actually. You can trick the airspeed selector module to detect an airspeed invalidity:
- set ASPD_DO_CHECKS to 1 to enable airspeed validation
- in flight, change ASPD_STALL to a very high number (e.g. 100m/s) --> as the current airspeed then is far below the set stall speed, it will declare the airspeed invalid and set it to NAN.

hamishwillee, 2020-03-16:
This needs to stay alive. Just got to find time for it.

hamishwillee, 2020-06-15:
This is still completely relevant and on my list.

NAmmann, 2022-07-08:
Hi all.
Some remarks or questions from my side:
Is it valid to say one failsafe overrides an other failsafe? When looking at the Note 3 for example it says that geofence failsafe can only be overriden by data link loss if failsafe is hold not rtl.
Maybe it is better to ask what flight mode has the highest priority?
Without having it tested i believe also geofence failsafe rtl would be overriden by a data link loss if the failsafe would be terminate. Am i right?

bozkurthan, 2022-07-10:
I need to jump back this document to reveal this problem. However, the countless merge operation is made. So, 3 years ago, your comments were true. But, now, we need to verify that override problem is still exists. Maybe, @hamishwillee  can help more than me. 

The issue is opened to decide which operation is more critical. I don't know if there is any work to improve failsafe action.

bkueng, 2022-10-13:
Thanks for your efforts @bozkurthan. I'm closing this as we can now simulate everything: https://docs.px4.io/main/en/config/safety_simulation.html

closed 2022-10-13 by bkueng

hamishwillee, 2022-10-25:
Thanks @bkueng - yes this can absolutely be closed. The fundamental point here was that it was not possible to resolve the behaviour of simultaneous/nested failsafes, so we would have to document the matrix of relationships.

Your change to set the rule as "the most severe response action is what gets done", and providing the simulator mean that it is possible for users to know with some certainty what will happen.

@bozkurthan As Beat says, thanks for this. And my apology that we never got it into the docs until no longer needed.

bozkurthan, 2022-10-26:
No worries. Same thought comes to my mind. Thanks for affords on failsafe action. Main goal is to provide detailed information. As you say there is no need for documentation since there is simulation that covers all of them.

Thanks for response :)
@bkueng and @hamishwillee