PX4/PX4-Autopilot #13723: camera trigger massages in mavlink

moreba1, 2019-12-10:
Hi

does px4 send camera trigger positions send by mavlink ? 
, i do not see camera signs in qgroundcontrol in survey mission mode .

julianoes, 2019-12-16:
I don't think `camera_trigger` sends the `CAMERA_IMAGE_CAPTURED` message. I guess this could/should be added to the mavlink module.

julianoes, 2019-12-16:
Should we add that @DanielePettenuzzo?

ndepal, 2019-12-17:
`camera_feedback` publishes `ORB_ID(camera_capture)` which is translated to the `CAMERA_IMAGE_CAPTURED` mavlink message. I don't know what QGC does/is supposed to do with this information.

julianoes, 2019-12-17:
`CAMERA_IMAGE_CAPTURED` would show a photo icon in QGC.

moreba1, 2019-12-17:
but i don't see photo icon in QGC. should we use feedback from camera hot shoe to see icons ?

moreba1, 2020-01-19:
@ndepal 
@julianoes 
https://github.com/mavlink/qgroundcontrol/issues/8216#issuecomment-575943525

Don Gange : "QGC will only show image capture points if CAMERA_IMAGE_CAPTURED.capture_result == 1 which indicates a successful capture according to mavlink spec. In the screen shown above you can see that CAMERA_IMAGE_CAPTURED.capture_result equals -1 which isn't even a valid value for that field. So as far as I can tell QGC is working correctly."

hamishwillee, 2020-04-24:
@julianoes Who should be assigned owner to this - either to close or progress?

julianoes, 2020-04-24:
@hamishwillee whoever has a setup and can make and test the required changes :man_shrugging:.

hamishwillee, 2020-04-27:
Well, it definitely needs to be fixed, so I have hit the pinned button.

closed 2020-09-07 by moreba1