PX4/PX4-Autopilot #13724: Once camera_capture on, no fmu pwm outputs

hyonlim, 2019-12-10:
**Describe the bug**
When a system is using FMU output as main PWM and camera_capture running, no motor spin when it is armed.

**To Reproduce**
Steps to reproduce the behavior:
1. Set FMU PWM as main output
2. Run camera_capture and on
3. Try to arm

**Expected behavior**
Motor should spin but it is not. If I do $ fmu test and arm again, it spins.

julianoes, 2019-12-16:
This looks like the usual configuration issue between output and input pins.

I think what you should try is to use something like `fmu mode_...` to configure `fmu` to only use and configure some of the pins and then run `camera_capture` on the remaining pins.

julianoes, 2020-03-20:
Anyone there?

closed 2020-03-20 by julianoes