PX4/PX4-Autopilot #13731: Kakute F7 - baro sensor #0 missing

sevet, 2019-12-11:
Failing prearm check with error: baro sensor #0 missing compiled from source on 11th dec 2019

**Steps to reproduce
Compile
Flash
Calibrate and setup all
Try to arm.

You can also see on the map in q ground control that the altitude is not changing when elevating the quad

**Expected behavior**
Altitude support

dagar, 2019-12-11:
Can you get boot output? From a console run `dmesg`.

dagar, 2019-12-11:
You could also try starting the barometer manually. `bmp280 -I start`

https://github.com/PX4/Firmware/blob/d6ffdc09b82f4ba9b922e4230a594d6bf50d5bc8/boards/holybro/kakutef7/init/rc.board_sensors#L15

sevet, 2019-12-16:
Hi, sorry for the late response, starting the bmp280 manually does not help, I also have problems saving config after reboot, here is the dmesg output and the saving error:

nsh> bmp280 -I start
nsh> dmesg
h>
nsh> nsh>
nsh> nsh> dmsg
nsh> nsh: dmsg: command not found
nsh> nsh> dmesg
nsh> sercon: Registering CDC/ACM serial driver
nsh> sercon: Successfully registered the CDC/ACM serial driver
nsh> HW arch: HOLYBRO_KAKUTEF7
nsh> FW git-hash: 944bf54d84815cb668aa5975fd5820386ebe14a7
nsh> FW version: 1.10.0 80 (17432704)
nsh> FW git-branch: master
nsh> OS: NuttX
nsh> OS version: Release 8.2.0 (134349055)
nsh> OS git-hash: 2e55ce8ec97d37b3d3c46b1aaab0a09364fcf254
nsh> Build datetime: Dec 11 2019 08:57:50
nsh> Build uri: localhost
nsh> Toolchain: GNU GCC, 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
nsh> PX4GUID: 000200000000363636313036510c00350037
nsh> MCU: STM32F74xxx, rev. Z
nsh> ERROR [param] importing failed (-1)
nsh> INFO  [tune_control] Publishing standard tune 1
nsh> Board defaults: /etc/init.d/rc.board_defaults
nsh>   SYS_AUTOSTART: curr: 0 -> new: 4050
nsh>   SYS_MC_EST_GROUP: curr: 2 -> new: 3
nsh>   ATT_ACC_COMP: curr: 1 -> new: 0
nsh>   ATT_W_ACC: curr: 0.2000 -> new: 0.4000
nsh>   ATT_W_GYRO_BIAS: curr: 0.1000 -> new: 0.0000
nsh>   SYS_HAS_MAG: curr: 1 -> new: 0
nsh>   CBRK_BUZZER: curr: 0 -> new: 782090
nsh> INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
nsh> ERROR [param] Parameter LIGHT_EN_BLINKM not found
nsh>   NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
nsh>   RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
nsh>   RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
nsh>   RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
nsh>   PWM_MAX: curr: 2000 -> new: 1950
nsh>   PWM_MIN: curr: 1000 -> new: 1075
nsh>   MC_ROLL_P: curr: 6.5000 -> new: 8.0000
nsh>   MC_ROLLRATE_P: curr: 0.1500 -> new: 0.0800
nsh>   MC_ROLLRATE_I: curr: 0.2000 -> new: 0.2500
nsh>   MC_ROLLRATE_D: curr: 0.0030 -> new: 0.0010
nsh>   MC_PITCH_P: curr: 6.5000 -> new: 8.0000
nsh>   MC_PITCHRATE_P: curr: 0.1500 -> new: 0.0800
nsh>   MC_PITCHRATE_I: curr: 0.2000 -> new: 0.2500
nsh>   MC_PITCHRATE_D: curr: 0.0030 -> new: 0.0010
nsh>   MC_YAW_P: curr: 2.8000 -> new: 4.0000
nsh>   MC_ROLLRATE_MAX: curr: 220.0000 -> new: 1600.0000
nsh>   MC_PITCHRATE_MAX: curr: 220.0000 -> new: 1600.0000
nsh>   MC_YAWRATE_MAX: curr: 200.0000 -> new: 1000.0000
nsh>   MPC_MANTHR_MIN: curr: 0.0800 -> new: 0.0000
nsh>   MPC_MAN_TILT_MAX: curr: 35.0000 -> new: 60.0000
nsh>   THR_MDL_FAC: curr: 0.0000 -> new: 0.3000
nsh>   PWM_RATE: curr: 400 -> new: 0
nsh>   SDLOG_PROFILE: curr: 3 -> new: 19
nsh>   RC_FLT_CUTOFF: curr: 10.0000 -> new: 0.0000
nsh>   CBRK_IO_SAFETY: curr: 0 -> new: 22027
nsh> Board sensors: /etc/init.d/rc.board_sensors
nsh> WARN  [mpu6000] no device on bus #3 (SPI4)
nsh> MPU6000 on SPI bus 4 at 1000 KHz)
nsh> INFO  [it8310] no device on be: 2
nsh> ERROR [hmc5883] bus 2ot started
nsh> ERROR [hmc5883]d (bus: 10 KHz, max: 400 KHz)
nsh> ERROR [fmu] setting I2Ctund
nsh> Boad extras: /etc/it.d/rc.board_mavlink
nsh> INFO  sn /dev/ttACM0 @ 57600B
nsh> Starting Main GPS on /tarting MAVLinn /dev/ttyS0
nsh> INFO  [mavlink] mode: NBINF [init] Mixer /etc/mixers/quad_x.main.mr MNTODE_IN not found
nsh> Board xtraer] logger sarted (moe=all)
nsh>
nsh> bmp280 -I start
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh> bmp280 -I start
nsh> dmesg
nsh> ERROR [parameters] param auto save failed (-27)

bkueng, 2019-12-17:
Fixed in https://github.com/PX4/Firmware/pull/13755.

closed 2020-03-12 by dagar