PX4/PX4-Autopilot #13732: Durandal Mission transfer fail

Tony3dr, 2019-12-11:
**Describe the bug**
We were able to flash master on the latest QGC 3.5.6 on the Durandal flight controller. The problem is that we are getting a Mission transfer fail error message when loading the Mission on QGC. We tried with USB and Mission failed, USB and battery connected and Mission failed, it seems to work via Telemetry. 

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure you have the latest QGC 3.5.6
2. Flash firmware Master branch
3. Uploaded mission to the vehicle
4. Error message "Mission transfer failed" see picture below.
[image]

**Log Files**
**Modes Tested**
Position Mode: Good.
Altitude Mode: Good.
Stabilized Mode: Good.
Mission Plan Mode (Automated): Good.RTL (Return To Land): Good.
https://review.px4.io/plot_app?log=0a193cd7-9b3b-483f-86de-aea71489b32b
https://review.px4.io/plot_app?log=62b6b784-7e3a-41c2-a47b-b27791a90635

**Drone Information**
Firmware: v1.10.0 (beta) (d6ffdc09)
Flight Controller: Durandal(Holybro)
GPS: Holybro GPS
Frame: F450
Telemetry: Holybro 915Mhz

davids5, 2019-12-12:
@Tony3dr - please add the list of what was connected and when.

Junkim3DR, 2019-12-19:
### Durandal Mission Transfer BenchTest.

**Tested with Mac OS**
- Bench Test, Uploading Mission to vehicle with USB Cable and Mac.
  - Log: https://review.px4.io/plot_app?log=bf2c4497-890b-404f-97b6-6274b7255ba1
- Bench Test, Uploading Mission to vehicle with USB Cable and Battery with Mac.
  - Log: https://review.px4.io/plot_app?log=006edc4c-b5be-44cc-9cdb-51bbb42e1283
- Bench Test, Uploading Mission to vehicle telemetry with Mac.
  - Log: https://review.px4.io/plot_app?log=ad297afb-ae35-4b01-ac42-cb8fcfa321f9

**Tested with Windows OS**
- Bench Test, Uploading Mission to vehicle with USB Cable and Mac.
  - Log: https://review.px4.io/plot_app?log=00ad8885-dba8-4370-bb2b-7ed76535801d
- Bench Test, Uploading Mission to vehicle with USB Cable and Battery with Windows.
  - Log: https://review.px4.io/plot_app?log=ad152767-7eb5-42b3-ab4b-75b9dbd30475
-  Bench Test, Uploading Mission to vehicle telemetry with Windows.
  - Log: https://review.px4.io/plot_app?log=31993e1d-19c4-4242-971d-aa3a4da4f64d

### Notes

QGC popped up this messages after every few tests.

[image]

[image]

Every time this error was shown, we turned off the vehicle, and tried again but the issue persisted, so we turned off the vehicle took the SD card out and placed it back in, after this "troubleshooting" step, we were able to by pass this error every time it popped up.

julianoes, 2020-01-30:
@Tony3dr is this still an issue?

Tony3dr, 2020-01-30:
We flashed the master branch on the latest QGC 3.5.6 on the Durandal flight controller. Same issue getting, Mission transfer fail error message when loading the Mission on QGC, via USB cable. We tried with USB and battery connected and Mission was uploaded successfully. However, we are encountering another error, Baro Sensor #0, see images below: 
[image]
[image]
[image]
[image]
@julianoes

julianoes, 2020-01-31:
@Tony3dr ok that's not good. :confused: Is this only on one hardware unit or consistent?

davids5, 2020-01-31:
@Tony3dr do you have a powered USB hub?

Can you test again, with a powered hub and no battery?
Wall plug--V
[PC]--------[HUB]------[DURANDAL]

Tony3dr, 2020-01-31:
> @Tony3dr ok that's not good. 😕 Is this only on one hardware unit or consistent?
@julianoes this is only with the only hardware unit we have. 

@davids5 
Wall plug--V
[PC]--------[HUB]------[DURANDAL]
Connected the PC to the HUB and to the Durandal and vehicle connected good, flashed the master firmware, uploaded the Mission to the vehicle successfully. Tried to arm the vehicle and getting the same error message: Preflight fail, Baro Sensor #0.

davids5, 2020-04-04:
@PX4/testflights - @Tony3dr  - please retest with https://github.com/PX4/Firmware/pull/14592 
USE USB only (No battery) 

Do 20 (Plug in - upload - unplug) cycles on bench. Record results

closed 2020-04-04 by dagar