PX4/PX4-Autopilot #13742: PIXHAWK  SD card startup ,parameter are missing your firmware has a bug on it  !!

SebastienFerrero, 2019-12-13:
Dear all, I'm kindly looking for your help to get this cute pixhawk working.
After loading the Simulink model to the pixhawk using the awsome 'Embedded Coder Support Package for PX4 Autopilots', I couldn't arm the motors and coudn't start the flight (I have no ability to control the pixhawk through the RC transmitter), I'm sure that there is something missing in the 'rc.txt' file in the SD card, but I do not know how to fix that.
Also I have the following errors;

[image]

I also tried manually to load the parameters but the problem persists:
param set COM_FLTMODE1	3
param set COM_FLTMODE2	1
....

here is my rc.txt:
#Copyright 2018 The MathWorks, Inc.

# This is the custom rc.txt which loads px4_simulink_app on start-up
usleep 1000
uorb start
usleep 1000
#nshterm /dev/ttyACM0 & #Disable the USB serial console
usleep 1000
tone_alarm start
usleep 3000
px4io start
gps start # #gps start -f Starts GPS driver and Fake a GPS signal (useful for testing)
usleep 1000
sh /etc/init.d/rc.sensors
usleep 1000
#Uncomment the below 2 lines to use LPE estimator
#attitude_estimator_q start
#local_position_estimator start
#Using EKF2 estimator by default as PX4 does build LPE on px4fmu-v2 due to a limited flash.
ekf2 start
usleep 1000
mtd start
usleep 1000
param load /fs/mtd_params
usleep 1000
rgbled start
usleep 1000
fmu mode_pwm # This is required for AUX PWM channels
usleep 1000
dev/ttyACM0 -m config –x
px4_simulink_app start
nshterm /dev/ttyS6 & #this will make ttyS6 the serial console
mavlink boot_complete

#exit #4-Jan-2019

Thank you

julianoes, 2019-12-16:
Is it possible that you updated PX4 without adapting your startup script to the changes?
What version of PX4 are you using?

SebastienFerrero, 2019-12-17:
Dear julianoes : Hello
Thanks for your replay and your time, this is really kind of you.
I'm using  this 
[image]

Kindly, Where can I find the appropriate rc.txt file.
I'm using MATLAB R2019b. and the rc.txt file is provided in C:\ProgramData\MATLAB\SupportPackages\R2019b\toolbox\target\supportpackages\px4\lib\etc

Dear julianoes , I would like to inform you that another freind suggested the following

{{Hi, 
try 'px4io safety_off' in the nsh or rc.txt. Maybe its just in save mode...
Best, Sebastian }}

But still not working

BESTS

julianoes, 2019-12-17:
Do you know with what version MATLAB has tested this? I really have not seen MATLAB connected to PX4 like this in years.

SebastienFerrero, 2019-12-18:
Hello @julianoes 
Really don't know. I found this in matlab site: https://www.mathworks.com/help/supportpkg/px4/ug/px4-system-startup-sdcard.html searchHighlight=Customize%20PX4%20System%20Startup&s_tid=doc_srchtitle

Note
Before customizing the PX4 system startup, ensure that the Pixhawk Series flight controller is flashed with PX4 firmware. If you were using ArduPilot firmware before, or if you are not sure about the current firmware type and version of the flight controller (for example, if it is a new Pixhawk flight controller board), flash the latest stable release of the PX4 firmware, using QGroundControl. Refer to QGroundControl documentation for more details on uploading the PX4 firmware. This action needs to be done only once, for a specific board.

BEST

julianoes, 2019-12-19:
Interesting, it doesn't have any date references or PX4 version information. It's hard to know with what state this was last tested. Could you ask the people behind this MATLAB package?

You can also try to flash 1.8 or 1.9 to see if it works with that.

SebastienFerrero, 2019-12-19:
Thank you @julianoes 
I see. I already asked them, still waiting for the answer, a week ago.
[image]

SebastienFerrero, 2019-12-19:
@julianoes , all that I know is that the Embedded coder support package uses the 1.8.0 firmware otherwise it doesn't work.

SebastienFerrero, 2019-12-28:
Hello @julianoes

Someone suggest this 
[image]

Does it mean anything to you?

SebastienFerrero, 2019-12-28:
The problem is that I can't load parameters while using sd card.
[image] 

need help 
thanks

julianoes, 2020-01-30:
Could you connect a [debug probe](https://dev.px4.io/v1.9.0/en/debug/system_console.html#connecting-via-dronecode-probe) and log the whole startup output?