PX4/PX4-Autopilot #13743: make after a board.h changes does not rebuild correctly and is HUGE time sink

davids5, 2019-12-13:
This is becoming a problem on HIGH cycle time debug-fix-build-download loops. 

Make a change build and download - still have the bug! Oh make did not work. 

FYI @dinomani

dagar, 2019-12-13:
It appears to be the header dependencies within NuttX.

davids5, 2019-12-13:
Can you fix it?

julianoes, 2020-03-26:
@davids5 can you help me understand which file you mean and where it is included in the build? Then I can check it.

davids5, 2020-03-26:
@julianoes - Thanks for looking into this. When working on a board and changing files in the nuttx-config directory: like https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v5/nuttx-config/include/board.h
OR
https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v5/nuttx-config/scripts/script.ld

When I filed this Nuttx was not rebuilt and PX4 was no re-linked.  - It may be fixed by now.

dagar, 2020-03-26:
Some of those cases we can fix effectively. Some of the ones that intersect with the generated path (copy nuttx, do some work, etc) are a bit more subtle with a long chain of dependencies. 

In trying to fix it more generally in https://github.com/PX4/Firmware/pull/13747 I kept breaking it one way or the other (too strict always rebuilding everything OR too loose occasionally not rebuilding all the way through the dependency graph). 

Overall in https://github.com/PX4/Firmware/pull/13747 I came to the conclusion that carrying a 35000 line commit for native cmake support (https://github.com/PX4/NuttX/compare/pr-cmake_master?expand=1) might not be that crazy...

julianoes, 2020-03-30:
@dagar I was digging into this a bit because I'm convinced it should be fixable.
However, now I found that the two files below are both used and I'm wondering if that's on purpose?

https://github.com/PX4/Firmware/blob/master/platforms/nuttx/NuttX/CMakeLists.txt
https://github.com/PX4/Firmware/blob/master/platforms/nuttx/cmake/init.cmake

dagar, 2020-03-30:
> @dagar I was digging into this a bit because I'm convinced it should be fixable.
> However, now I found that the two files below are both used and I'm wondering if that's on purpose? [...]

Yes it's on purpose. At configure time we need to copy NuttX into the build directory and inflate the compressed defconfig so that the flags can be available to cmake.

Example: https://github.com/PX4/Firmware/blob/73f7243bc8350664864685d1a0adbeaaa2f2c5e5/platforms/nuttx/cmake/px4_impl_os.cmake#L79

At configure time these steps are done with execute_process(). Then we also want to catch any changes later for incremental builds, so most of the logic is duplicated as custom commands as well. 

Yes it's horrible, and one more little detail that factors into my comment (https://github.com/PX4/Firmware/issues/13743#issuecomment-604440859). 

If NuttX supported out of tree builds (by moving to cmake or even with the existing build system) it would simplify or solve many problems.

julianoes, 2020-03-31:
I see. I would think the copying needs to happen when a file changes and not at configure time, and that's probably partly where this issue comes from.

davids5, 2020-03-31:
@julianoes - So the complete copy is in the configure? 

think any change in nuttx-config to needs force the configure and do a copy of the changed files. 

This covers defconfig cases that change the configure step and file changes that effect the NuttX build.

dagar, 2020-03-31:
> I see. I would think the copying needs to happen when a file changes and not at configure time, and that's probably partly where this issue comes from.

It has to be copied initial at configure time for NuttX to do its own configuration, then the dependencies have to be in place to recopy and reconfigure as needed for incremental changes.

It's this relatively long chain of dependencies where things are breaking down (https://samthursfield.wordpress.com/2015/11/21/cmake-dependencies-between-targets-and-files-and-custom-commands/). I need to refresh the details, but I believe there's more subtle detail beyond that helpful blog post. I vaguely recall the issue was something to do with the long chain of dependencies ending up in separate generated Makefiles and then parallel make would run the earlier shared steps (eg the initial copy or configure) multiple times. It might be easier to talk through with an example.

julianoes, 2020-04-01:
I was exactly looking for such dependency problems and I could not even see anything wrong, at least in https://github.com/PX4/Firmware/blob/master/platforms/nuttx/NuttX/CMakeLists.txt but somehow nothing really had an effect.

I'm planning to look more in-depth at all the pieces to understand it and then I'm hoping to be able to boil it down to where it fails, maybe something for the weekend...