PX4/PX4-Autopilot #13744: Large yaw drift while stationary

aalexjo, 2019-12-13:
During testing i would frequently encounter the critical warning: Preflight fail: yaw estimate error. After looking a bit at the yaw while the imu is stationary, this is quite understandable. The yaw looks fairly stable until an initial disturbance, after which is drifts significantly. I have played around with the EKF parameters and not have any success. This problem is quite similar to #11847, but i am trying to use the magnometer to constrain the drift. The problem presents itself similarly regardless of the value of EKF2_MAG_TYPE.
[image]

Any suggestions, I would expecet it to be a hardware issue, but the IMU and MAG values seem normal. No GPS is used, though i wouldnt expect that to cause these types of issues.

julianoes, 2019-12-16:
I suggest that you set `SDLOG_MODE` to create a log from boot and then paste the log here. The plot above is not really enough for me to grasp what is going on.

Also, a picture of your setup might help.

aalexjo, 2019-12-16:
Hi thanks ill attavh a log file here where i initially disturb at around 15 seconds and leave it stationary for the next 4 minutes and then disturbe it again. The external disturbances are fairly easy to recognize. 

I have tried to narrow down the problem a bit so will attach some more plots here. 
It seems like the problem has somthing to do woth the fusion of the IMU data as you can see the HIGHRES_IMU drifts quite a bit after a disturbance, while the SCALED_IMU gyro values look to be quite accurate. 

These plots are with all the default values in the EKF2, but i have tried tweeking the time constants and process noise variables without really producing any different results.

Here is the plot of the HIGHRES_IMU
[image]

and here is the SCALED_IMU data for bith IMU's

[image]

Another image where i try to ilustrate the problem. 4 external disturbances were applied
t1: small motion with no absolute rotation
t2: Small rotation of the MAV
t3: Larger rotation of MAV
t4: High frequency oscillation of yaw.
[image]

During startup all values are stable as well as after a long period the yaw will eventually stabilize.

[log_79_2019-12-16-17-01-40.ulg.tar.gz](https://github.com/PX4/Firmware/files/3969012/log_79_2019-12-16-17-01-40.ulg.tar.gz)

julianoes, 2019-12-17:
Please upload the log file to https://logs.px4.io, thanks.

aalexjo, 2019-12-17:
https://logs.px4.io/plot_app?log=45acb2b4-448b-4b20-92e5-d9f087e30d4e

julianoes, 2019-12-19:
Thanks for the log @aalexjo. That's interesting and I don't think I understand what is going on.

@priseborough would you mind having a quick look at the above log to see if something springs to mind? Thanks.

bresch, 2019-12-25:
@aalexjo Do you have a picture of your setup? Are you sure that the mag rotation is correct?

aalexjo, 2019-12-25:
Unfortunately i dont have a very revealing picture of my setup handy, nor the drone. But the mag rotation might be onto something.  I have de-cased the pixhawk 4 mini and softmounted the IMU directly on the flight controller. This involved rotating the IMU unit 180 degrees and mounting it on the bottom of the FCU and mounting this setup so that the IMU was mounted in the standard orientation. I¨ve assumed that the mag was co-located on this unit, but if that is not the case then the  IMU would be flipped 180 degrees in roll compared to the mag which might explain the failure to estimate the bias. I am not sure if this is the case though, i will do some tests again when i have the equipment in front of me.

In the meantime can anyone confirm that this might be the problem?