PX4/PX4-Autopilot #13746: Main Motor Output Before Arming

nbelanger99, 2019-12-13:
**Describe the bug**
Using the 16001 Heli mixer on 1.9, the FMU outputs to MAIN5 (the tail rotor) before arming. The output will start at 1110 uS, and then after a few seconds to about 1750 uS and stay there until Armed, but will change as I physically yaw the controller. PWM_DISARMED is set to the default, 900, and all PWM_MAIN_DIS are set to the default -1. When Armed, the output is as expected

**To Reproduce**
Steps to reproduce the behavior:
1. Use the 16001 airframe
2. Scope MAIN5 (or connect to an ESC or servo or something)
3. Power Cycle the Autopilot
4. See error

**Expected behavior**
There should be no output, as with the other MAIN outputs

Before Arming (after a few seconds of 1110 uS output):
[image]

After arming- tail rotor (blue) and main rotor (yellow) are at minimum as expected
[image]

**Hardware:**
- Conventional Heli with tail rotor
- Pixhawk 4

**Note**
The minimums for both motors when armed is not exactly the same as the default mixer; we've adjusted it a bit for out aircraft, but the behavior is seem both with the modified and unmodified mixer file.

nbelanger99, 2019-12-13:
Console Output

 6c6bc46ec2434f94c9cc535417fe1a17febad7a9
FW version: 1.9.0 0 (17367040)
FW git-branch: 230
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: d8da511082646d83a54c6905daca13f0a1a609f0
Build datetime: Nov 11 2019 15:35:55
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0002000000003533383731385114002d0024
MCU: STM32F76xxx, rev. Z
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [px4_work_queue] creating: wq:lp_default, priority: 205, stack: 1700 bytes
INFO  [px4_work_queue] creating: wq:I2C1, priority: 248, stack: 1250 bytes
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [px4_work_queue] creating: wq:hp_default, priority: 243, stack: 1500 bytes
Board sensors: /etc/init.d/rc.board_sensors
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
INFO  [px4_work_queue] creating: wq:SPI1, priority: 254, stack: 1400 bytes
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
INFO  [ist8310] no device on bus 1
INFO  [px4_work_queue] creating: wq:I2C2, priority: 247, stack: 1250 bytes
INFO  [ist8310] no device on bus 2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 4 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
INFO  [px4_work_queue] creating: wq:I2C3, priority: 246, stack: 1250 bytes
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
INFO  [px4_work_queue] creating: wq:SPI5, priority: 250, stack: 1400 bytes
ERROR [pmw3901] driver start failed
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
INFO  [px4_work_queue] creating: wq:SPI4, priority: 251, stack: 1400 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 244, stack: 7000 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 255, stack: 1600 bytes
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/blade130.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> [KINFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: /fs/microsd/log/2019-12-13/04_13_56.ulg
INFO  [logger] closed logfile, bytes written: 119379
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: /fs/microsd/log/2019-12-13/04_14_04.ulg
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 525221
WARN  [commander] Manual control lost
ERROR [mavlink] instance 1: RADIO_STATUS timeout
[... 192 lines cut ...]
ist8310_conf_err: 0 events
ist8310_rng_err: 0 events
ist8310_com_err: 0 events
ist8310_read: 236 events, 204962us elapsed, 868.48us avg, min 839us max 1627us 1234.721us rms
bmi055_gyro_bad_registers: 0 events
bmi055_gyro_bad_transfers: 0 events
bmi055_gyro_measure_interval: 6256 events, 799.87 avg, min 520us max 1083us 26.085us rms
bmi055_gyro_read: 6256 events, 226770us elapsed, 36.25us avg, min 33us max 58us 45.585us rms
bmi055_accel_duplicates: 1008 events
bmi055_accel_bad_registers: 0 events
bmi055_accel_bad_transfers: 0 events
bmi055_accel_measure_interval: 6257 events, 799.87 avg, min 536us max 1028us 53.911us rms
bmi055_accel_read: 6257 events, 190673us elapsed, 30.47us avg, min 23us max 48us 64.963us rms
mpu6k_duplicates: 1257 events
mpu6k_reset: 0 events
mpu6k_bad_reg: 0 events
mpu6k_bad_trans: 0 events
mpu6k_measure_interval: 6257 events, 799.87 avg, min 527us max 1073us 28.920us rms
mpu6k_read: 6257 events, 495853us elapsed, 79.25us avg, min 37us max 251us 986.428us rms
stack_check: 42 events, 145us elapsed, 3.45us avg, min 1us max 33us 99.471us rms
dma_alloc: 0 events
param_set: 1 events, 14us elapsed, 14.00us avg, min 14us max 14us   infus rms
param_get: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param_find: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param_export: 1 events, 40310us elapsed, 40310.00us avg, min 40310us max 40310us   nanus rms

bresch, 2019-12-17:
Do you have a log file from boot to share? The "tail" output is supposed to be connected to a servo and not a motor, it is then expected to run in prearm state. Are the swashplate servos also moving or not?

nbelanger99, 2019-12-17:
> Do you have a log file from boot to share? The "tail" output is supposed to be connected to a servo and not a motor, it is then expected to run in prearm state. Are the swashplate servos also moving or not?

1. No, the (other) servos have no output when not armed.
2. Why should the tail be treated as a servo? The Blade 130, as our heli, has a conventional tail rotor. 

Here's a log that started at boot; I armed and disarmed twice:

```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: 6c6bc46ec2434f94c9cc535417fe1a17febad7a9
FW version: 1.9.0 0 (17367040)
FW git-branch: 230
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: d8da511082646d83a54c6905daca13f0a1a609f0
Build datetime: Dec 17 2019 14:29:32
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0002000000003533383731385114002d0024
MCU: STM32F76xxx, rev. Z
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [px4_work_queue] creating: wq:lp_default, priority: 205, stack: 1700 bytes
INFO  [px4_work_queue] creating: wq:I2C1, priority: 248, stack: 1250 bytes
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
INFO  [px4_work_queue] creating: wq:hp_default, priority: 243, stack: 1500 bytes
Board sensors: /etc/init.d/rc.board_sensors
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
INFO  [px4_work_queue] creating: wq:SPI1, priority: 254, stack: 1400 bytes
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
INFO  [ist8310] no device on bus 1
INFO  [px4_work_queue] creating: wq:I2C2, priority: 247, stack: 1250 bytes
INFO  [ist8310] no device on bus 2
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
WARN  [qmc5883] no device on bus 1 (type: 2)
WARN  [qmc5883] no device on bus 2 (type: 2)
WARN  [qmc5883] no device on bus 4 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
WARN  [lis3mdl] no device on bus 2
INFO  [px4_work_queue] creating: wq:I2C3, priority: 246, stack: 1250 bytes
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
INFO  [px4_work_queue] creating: wq:SPI5, priority: 250, stack: 1400 bytes
ERROR [pmw3901] driver start failed
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
INFO  [px4_work_queue] creating: wq:SPI4, priority: 251, stack: 1400 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 244, stack: 7000 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 255, stack: 1600 bytes
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/blade130.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
[... 135 lines cut ...]
ist8310_rng_err: 0 events
ist8310_com_err: 0 events
ist8310_read: 2087 events, 1811455us elapsed, 867.97us avg, min 838us max 1712us 62.909us rms
bmi055_gyro_bad_registers: 0 events
bmi055_gyro_bad_transfers: 0 events
bmi055_gyro_measure_interval: 55375 events, 799.99 avg, min 546us max 1057us 11.129us rms
bmi055_gyro_read: 55375 events, 2006610us elapsed, 36.24us avg, min 33us max 56us 2.644us rms
bmi055_accel_duplicates: 8842 events
bmi055_accel_bad_registers: 0 events
bmi055_accel_bad_transfers: 0 events
bmi055_accel_measure_interval: 55376 events, 799.99 avg, min 416us max 1163us 56.307us rms
bmi055_accel_read: 55376 events, 1683016us elapsed, 30.39us avg, min 23us max 52us 3.641us rms
mpu6k_duplicates: 11182 events
mpu6k_reset: 0 events
mpu6k_bad_reg: 0 events
mpu6k_bad_trans: 0 events
mpu6k_measure_interval: 55376 events, 799.99 avg, min 692us max 906us 7.479us rms
mpu6k_read: 55376 events, 4377374us elapsed, 79.05us avg, min 37us max 474us 53.710us rms
stack_check: 190 events, 976us elapsed, 5.14us avg, min 1us max 104us 8.696us rms
dma_alloc: 0 events
param_set: 4 events, 32us elapsed, 8.00us avg, min 2us max 13us 106.454us rms
param_get: 326 events, 1382us elapsed, 4.24us avg, min 2us max 80us 49.464us rms
param_find: 230 events, 1110us elapsed, 4.83us avg, min 2us max 75us 17.642us rms
param_export: 2 events, 98340us elapsed, 49170.00us avg, min 45884us max 52456us 4647.115us rms
```

nbelanger99, 2019-12-18:
Update:

I've found that this is happening only in the prearm state- that is, there is an output when CBRK_IO_SAFETY is set to defatult and the switch has been pressed, or when the switch is not present and CBRK_IO_SAFETY is set to 22027. 

This state logic happens in Commander.cpp, but I've tried in mixer.cpp, fmu.cpp, and px4io.cpp to manually override the output in order to understand the heirarchy of control here, but to no avail. The results remain.

hermankolden, 2022-04-04:
@nbelanger99 I had similar issues on PX4 v1.12.3, where the pusher motor on one (and only one) occasion started spinning immediately after disengaging the safety switch. Did you ever identify the cause or a solution?