PX4/PX4-Autopilot #13748: Sign Pixhawk board with COA

Javad97, 2019-12-14:
How to generate auth.c to sign pixhawk board
reference to this documentation https://ardupilot.org/dev/docs/how-to-use-the-auth-command-to-sign-a-pixhawk-board-with-your-certificate-of-authenticity.html#firmware

Thanks!

julianoes, 2019-12-16:
Are you using PX4 or ArduPilot?

Javad97, 2019-12-17:
Iam using ardupilot.

julianoes, 2019-12-17:
Ok, in that case I suggest that you ask in the [ArduPilot forums](https://discuss.ardupilot.org/) or on the [ArduPilot GitHub](https://github.com/ArduPilot/ardupilot). I don't think PX4 has much to do with this.

closed 2019-12-17 by julianoes

Javad97, 2019-12-17:
Actually I am a beginner, so I choose ardupilot although I didn't know about the auth signing commands.
can you explain to me how can it done with px4 firmware?

julianoes, 2019-12-19:
I don't think we're doing/using that right now.

What are you trying to achieve?

Javad97, 2019-12-19:
I want to assign a public/private key for the flight control board.

julianoes, 2019-12-19:
Taking a step back, what problem are you trying to solve?