PX4/PX4-Autopilot #13752: Kakute F7 - Not saving params

sevet, 2019-12-16:
**Describe the bug**
Any change to config is not restored after power cycle or reboot

**To Reproduce**
1. connect with qground control
2. make a config change like "sys_has_mag"
3. save & reboot
4. power on and see change was not saved.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
from console:
nsh>
nsh> ERROR [parameters] param auto save failed (-27)

bkueng, 2019-12-17:
Can you flash master and check the output of `dmesg`?

sevet, 2019-12-17:
Out put from a fresh build from today's master:

nsh> ERROR [parameters] param auto save failed (-27)
ERROR [parameters] param auto save failed (-27)

nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh>
nsh> dmesg
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: HOLYBRO_KAKUTEF7
FW git-hash: 7abadabc19ddc8c4c64931a56d7d7cb9a77d11d3
FW version: 1.11.0 80 (17498240)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 9331fda0d7f7220de33982e8a641c671b1f09c8e
Build datetime: Dec 17 2019 08:40:18
Build uri: localhost
Toolchain: GNU GCC, 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
PX4GUID: 000200000000363636313036510c00350037
MCU: STM32F74xxx, rev. Z
ERROR [param] importing failed (-1)
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
  SYS_AUTOSTART: curr: 0 -> new: 4050
  SYS_MC_EST_GROUP: curr: 2 -> new: 3
  ATT_ACC_COMP: curr: 1 -> new: 0
  ATT_W_ACC: curr: 0.2000 -> new: 0.4000
  ATT_W_GYRO_BIAS: curr: 0.1000 -> new: 0.0000
  SYS_HAS_MAG: curr: 1 -> new: 0
  CBRK_BUZZER: curr: 0 -> new: 782090
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
ERROR [param] Parameter LIGHT_EN_BLINKM not found
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  MC_ROLL_P: curr: 6.5000 -> new: 8.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.0800
  MC_ROLLRATE_I: curr: 0.2000 -> new: 0.2500
  MC_ROLLRATE_D: curr: 0.0030 -> new: 0.0010
  MC_PITCH_P: curr: 6.5000 -> new: 8.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.0800
  MC_PITCHRATE_I: curr: 0.2000 -> new: 0.2500
  MC_PITCHRATE_D: curr: 0.0030 -> new: 0.0010
  MC_YAW_P: curr: 2.8000 -> new: 4.0000
  MC_ROLLRATE_MAX: curr: 220.0000 -> new: 1600.0000
  MC_PITCHRATE_MAX: curr: 220.0000 -> new: 1600.0000
  MC_YAWRATE_MAX: curr: 200.0000 -> new: 1000.0000
  MPC_MANTHR_MIN: curr: 0.0800 -> new: 0.0000
  MPC_MAN_TILT_MAX: curr: 35.0000 -> new: 60.0000
  THR_MDL_FAC: curr: 0.0000 -> new: 0.3000
  PWM_RATE: curr: 400 -> new: 0
  SDLOG_PROFILE: curr: 3 -> new: 19
  RC_FLT_CUTOFF: curr: 10.0000 -> new: 0.0000
  CBRK_IO_SAFETY: curr: 0 -> new: 22027
Board sensors: /etc/init.d/rc.board_sensors
WARN  [mpu6000] no device on bus #3 (SPI4)
MPU6000 on SPI bus 4 at 0 (1000 KHz)
INFO  [ist8310] no device on bus 1
WARN  [hmc5883] no device on bus 1 (type: 2)
ERROR [hmc5883] bus 2 not started
ERROR [hmc5883] bus 2 not started
QMC5883_I2C on I2C bus 1 at 0x0d (bus: 100 KHz, max: 400 KHz)
ERROR [fmu] setting I2C clock failed
ERROR [param] Parameter TRIG_MODE not found
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting RC Input Driver on /dev/ttyS4
Starting MAVLink on /dev/ttyS0
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS0 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
ERROR [param] Parameter MNT_MODE_IN not found
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)

bkueng, 2019-12-19:
Can you try these commands?
```
param reset
param save
```

sevet, 2019-12-23:
> param reset

This seems to stop the error but doesn't seem to save params

julianoes, 2020-01-30:
@sevet has this been resolved?

julianoes, 2020-04-29:
No response, closing.

closed 2020-04-29 by julianoes

kh4lilo, 2023-07-12:
hello, i am having the same issue with the kakute H7, some params just refuse to be saved (PWM rate for example). any news abt this?

julianoes, 2023-07-12:
@kh4lilo can you attach your param file and PX4 version? It could be that the PWM param is overwritten by your airframe (SYS_AUTOSTART) choice.