PX4/PX4-Autopilot #13754: LAND commanded but not land.

hyonlim, 2019-12-17:
**Describe the bug**
Land was commanded, but vehicle is not landing at all.

**To Reproduce**
Vehicle was hit on the wall, EKF lost a little bit. Then, land is keep commanded, but never land.
Vehicle was drifted along wind direction. And later, battery was drained, it finally landed.

**Expected behavior**
Land immediately.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=2b373434-6c98-4c93-bbc3-675000b4725d

**Drone (please complete the following information):**
Firmware is bit old version. 6e395fe885740f88aa633ce747fb12e950110cb0.
However, I think this is not an issue.

julianoes, 2019-12-17:
I think this is the same as #11468 and should be fixed with #13143.

julianoes, 2020-03-20:
Closing as there was no follow-up.

closed 2020-03-20 by julianoes