PX4/PX4-Autopilot #13756: make make px4_sitl_rtps fails to find fastrtpsgen in default location

deb0ch, 2019-12-17:
**Describe the bug**
After installing dependencies using the [official script](https://raw.githubusercontent.com/PX4/Devguide/v1.9.0/build_scripts/ubuntu_sim.sh), running `make px4_sitl_rtps` fails saying that `fastrtpsgen` (which is installed in the default location `/usr/local/bin`) is not found, whereas [this is not supposed to be the case](https://dev.px4.io/v1.9.0/en/middleware/micrortps.html#agent-not-builtfastrtpsgen-is-not-found). 

Digging up the code, I found out that if `FASTRTPSGEN_DIR` is not set before _the first time_ that `make px4_sitl_rtps` is run, then the wrong command is issued by cmake with an empty `--fastrtpsgen-dir` option (`src/modules/micrortps_bridge/micrortps_client/CMakeLists.txt:74`).

According to the docs, you should not need the environment variable if fastRTPS is installed in the default location. Moreover, `make` states that it found `fastrtpsgen` with the correct version.

A sensible fix would be to hard-code a default value of `/usr/local/bin` in the CMakeLists.txt.

**To Reproduce**
Steps to reproduce the behavior:
1. Install dependencies using the [official script](https://raw.githubusercontent.com/PX4/Devguide/v1.9.0/build_scripts/ubuntu_sim.sh)
2. Build the sitl_rtps variant: `make px4_sitl_rtps`
4. See error

**Expected behavior**
No error, successful build.

**Log Files and Screenshots**
```
-- PX4 version: v1.11.0-beta1-3-g7abadabc19
-- PX4 config file: /home/deb0ch/src/PX4Firmware/boards/px4/sitl/rtps.cmake
-- PX4 config: px4_sitl_rtps
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- fastrtpsgen version 1.7.1
-- RTPS send: home_position, wind_estimate, radio_status, satellite_info, adc_report, iridiumsbd_status, vtol_vehicle_status, airspeed, sensor_selection, cpuload, sensor_baro, battery_status, estimator_status, distance_sensor, vehicle_odometry, vehicle_attitude, collision_constraints, sensor_combined, 
-- RTPS receive: trajectory_waypoint, position_setpoint, camera_capture, collision_report, obstacle_distance, debug_array, debug_key_value, debug_vect, position_setpoint_triplet, debug_value, camera_trigger, vehicle_trajectory_waypoint, optical_flow, vehicle_mocap_odometry, vehicle_visual_odometry
-- Configuring done
-- Generating done
-- Build files have been written to: /home/deb0ch/src/PX4Firmware/build/px4_sitl_rtps
[11/771] git submodule src/drivers/gps/devices
[23/771] Generating RTPS topic bridge
FAILED: src/modules/micrortps_bridge/micrortps_client/micrortps_agent/home_position_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/home_position_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/wind_estimate_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/wind_estimate_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/radio_status_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/radio_status_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/satellite_info_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/satellite_info_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/adc_report_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/adc_report_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/iridiumsbd_status_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/iridiumsbd_status_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vtol_vehicle_status_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vtol_vehicle_status_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/airspeed_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/airspeed_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_selection_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_selection_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/cpuload_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/cpuload_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_baro_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_baro_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/battery_status_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/battery_status_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/estimator_status_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/estimator_status_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/distance_sensor_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/distance_sensor_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vehicle_odometry_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vehicle_odometry_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vehicle_attitude_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/vehicle_attitude_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/collision_constraints_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/collision_constraints_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_combined_Publisher.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_combined_Publisher.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/trajectory_waypoint_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/trajectory_waypoint_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/camera_capture_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/camera_capture_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/collision_report_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/collision_report_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/obstacle_distance_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/obstacle_distance_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_array_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_array_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_key_value_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_key_value_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_vect_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_vect_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/position_setpoint_triplet_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_value_Subscriber.cpp src/modules/micrortps_bridge/micrortps_client/micrortps_agent/debug_value_Subscriber.h src/modules/micrortps_bridge/micrortps_client/micrortps_agent/camera_trigger_Subscriber.cpp [... cut ...] _bridge.log >/dev/null
Traceback (most recent call last):
  File "/home/deb0ch/src/PX4Firmware/msg/tools/generate_microRTPS_bridge.py", line 261, in <module>
    "FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag")
Exception: FastRTPSGen not found. Specify the location of fastrtpsgen with the -f flag
[29/771] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_sitl_rtps' failed
make: *** [px4_sitl_rtps] Error 1
```

julianoes, 2019-12-19:
FYI @TSC21

stmoon, 2020-01-01:
Hi,

I had same problem and solve it like #13822.

I hope it helps you :-)

deb0ch, 2020-01-03:
I mean, I solved the problem for myself, just something needs to be done for the general user, for the code to behave as expected and as the docs say

deb0ch, 2020-01-03:
I solved this for myself but pointed out that something needed to be done for the general user, for the code to behave as expected and as the docs say

pratikiitgn, 2021-04-06:
@deb0ch and @stmoon 

can you please help me to sort out this issue. I stuck here.

adk4davsys, 2021-05-20:
@deb0ch Could you please provide your solution to your own question you posted?

TSC21, 2021-05-20:
> @deb0ch Could you please provide your solution to your own question you posted?

@adk4davsys what problem are you facing? Did you follow the documentation in this regards? - https://docs.px4.io/master/en/dev_setup/fast-dds-installation.html.

adk4davsys, 2021-05-20:
> > @deb0ch Could you please provide your solution to your own question you posted?
> 
> @adk4davsys what problem are you facing? Did you follow the documentation in this regard? - https://docs.px4.io/master/en/dev_setup/fast-dds-installation.html.

@TSC21 Thank you for your reply. I followed along the doc up to the point where I issue the command: "source build_ros2_workspace.bash" and the terminal closes by itself after printing "fastrtpsgen not found."

I am using Ubuntu 20.04.

Please let me know what else I could try to get the offboard working with ros2 and PX4 when you get a chance.

Thanks!

nfry321, 2021-09-09:
I also had the problem where the command `source build_ros2_workspace.bash` caused the terminal to close during the build process - I was unable to see any error message before this though. 

To solve it I ran the script without `source` and had no further issues - i.e. `./build_ros2_workspace.bash`

xiake-11, 2021-09-23:
> I also had the problem where the command `source build_ros2_workspace.bash` caused the terminal to close during the build process - I was unable to see any error message before this though.
> 
> To solve it I ran the script without `source` and had no further issues - i.e. `./build_ros2_workspace.bash`

thank you very much, I have the same problem . I do as you say，but the result is printed out 
“./build_ros2_workspace.bash: line 37: fastrtpsgen: command not found” , could you help me

nfry321, 2021-09-23:
Have you added `fastrtpsgen` to your `PATH`? 
See: https://docs.px4.io/master/en/dev_setup/fast-dds-installation.html#fast-rtps-gen

deb0ch, 2021-11-20:
@adk4davsys  @pratikiitgn Sorry for responding so late, I guess I haven't been so attentive to my Github notifications for the past few years :sweat_smile: 

I don't remember for sure, but seeing the issue again I think that I most likely solved it by setting the `FASTRTPSGEN_DIR` variable to the location of my `fastrtpsgen` install (not sure which folder but surely one of them).

In any case, this should be solved by @stmoon 's PR https://github.com/PX4/PX4-Autopilot/pull/13822 no ? If it is the case then this issue should be closed.