PX4/PX4-Autopilot #13759: Bug of "make px4_sitl gazebo"

Miranda-Tian, 2019-12-18:
The version of my gazebo is 7.16.0.
First I `source ubuntu_sim_common_deps.sh`. Then I ran ` make px4_sitl gazebo`.
**Describe the bug**
The bug shown in terminal is in the following:
```
-- Configuring incomplete, errors occurred!
See also "/home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[8/505] Building CXX object platforms/...iles/px4_work_queue.dir/WorkItem.cpp.o
FAILED: cd /home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/parallels/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/parallels/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[8/505] Building CXX object platforms/...iles/px4_work_queue.dir/WorkItem.cpp.o
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```
And the content of CMakeError.log is below:
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/ninja" "cmTC_11f9e"
[1/2] Building C object CMakeFiles/cmTC_11f9e.dir/CheckSymbolExists.c.o
[2/2] Linking C executable cmTC_11f9e
FAILED: : && /usr/bin/cc     CMakeFiles/cmTC_11f9e.dir/CheckSymbolExists.c.o  -o cmTC_11f9e   && :
CMakeFiles/cmTC_11f9e.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

File /home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/parallels/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/ninja" "cmTC_316db"
[1/2] Building C object CMakeFiles/cmTC_316db.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_316db
FAILED: : && /usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTC_316db.dir/CheckFunctionExists.c.o  -o cmTC_316db  -lpthreads && :
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

It seems that there're some problems in 'pthread_create'. But I have no idea how to fix it. 
Can someone tell me how to fix that? Thanks!

julianoes, 2019-12-19:
Can you try `make clean` and then paste the output of `git status` and `make px4_sitl`?

Mavrikant, 2019-12-20:
I'm using `make  px4_sitl_default gazebo` and it works. But I'm not sure is it correct mentod.

Tarrandurham, 2019-12-21:
I'm having the same issue. Tried everything, reininstalled everything, buddy's system shows the same error. make clean, git status, make px_sitl - all done.

```
git status

On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

```

`sudo make px4_sitl
-- PX4 version: v1.11.0-beta1-28-g49cb21016a
-- PX4 config file: /home/parallels/px4/src1/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/px4/src1/Firmware/build/px4_sitl_default
[20/726] git submodule src/drivers/gps/devices
[21/726] git submodule src/lib/ecl
[27/726] git submodule mavlink/include/mavlink/v2.0
[726/726] Linking CXX shared library s...s/dyn_hello/examples__dyn_hello.px4mod`

piradata, 2020-02-02:
Same thing in there, when i run "no_sim=1 make px4_sitl_default gazebo" there is the following error:

```
[49/96] Building CXX object CMakeFiles...n.dir/src/gazebo_catapult_plugin.cpp.o
ninja: build stopped: subcommand failed.
[705/709] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/piradata/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
```

hamishwillee, 2020-02-02:
I just tested this off latest firmware environment set up as described in our [docs](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#sim_nuttx). This is Gazebo 9 on Ubuntu 18.04. For me `NO_SIM=1 make px4_sitl_default gazebo` works fine. 

I am closing this as the follow on posts are different reported errors as the original one.

@piradata Can you please try again with recent firmware and if you still have the problem report a new bug with detail on your environment setup. Perhaps create a gist with your full build log and link to it?

closed 2020-02-02 by hamishwillee

piradata, 2020-02-02:
okay, I will test it soon, now I am using the old version of px4 on my pc,
will format the laptop for a clean install

alithethird, 2020-05-04:
Had the same issues and re-downloaded gazebo plug-ins
`sudo apt-get install libprotobuf-dev libprotoc-dev protobuf-compiler libeigen3-dev libxml2-utils python-rospkg python-jinja2`
`sudo apt-get install libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly -y`

and it worked like a charm

Babyboomu, 2024-07-08:
Had the same issue but it still can't run in the correct way.

FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /home/mumu/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic &amp;&amp; /usr/bin/cmake --build /home/mumu/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 8
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1