PX4/PX4-Autopilot #13762: pwm rates can not be set more than 400hz

feemi, 2019-12-18:
**Describe the bug**
pwm rate parameters range is -1 to 2000 however you can not set more than 400hz. It sets to 400 if you try to set more than this value.

**To Reproduce**
Steps to reproduce the behavior:
1. open mavlink console
2. run `pwm rate -r 600 -c 1234 `command
    
3.  run `pwm info` command
4. See values limited to 400 hz

**Expected behavior**
pwm rates should be set as desired.

**Log Files and Screenshots**
[image]
[image]

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- pixhawk cube 2.1 fmu v3
- px4 v1.9.2

feemi, 2019-12-18:
It seems that solution can apply this chunk of code in registers.c file
But I dont think this values should be hardcoded like this.
Is it okay to add min max values to  drv_pwm_output.h ?

`		case PX4IO_P_SETUP_PWM_DEFAULTRATE:
			if (value < 25) {
				value = 25;
			}

			if (value > 400) {
				value = 400;
			}

			pwm_configure_rates(r_setup_pwm_rates, value, r_setup_pwm_altrate);
			break;

		case PX4IO_P_SETUP_PWM_ALTRATE:

			/* For PWM constrain to [25,400]Hz
			 * For Oneshot there is no rate, 0 is therefore used to select Oneshot mode
			 */
			if (value != 0) {
				if (value < 25) {
					value = 25;
				}

				if (value > 400) {
					value = 400;
				}
			}

			pwm_configure_rates(r_setup_pwm_rates, r_setup_pwm_defaultrate, value);
			break;
`

mwiatt, 2019-12-18:
The standard in RC PWM control is a 1ms pulse indicating "low" and a 2 ms pulse indicating "high". A 400 Hz update rate gives a 2.5 ms period. Anything much more than a 400 Hz update rate would not allow a 2ms pulse to get through. That is why the PWM refresh rate is capped at 400 Hz.

FlavioTonelli, 2021-06-23:
I'm right now facing the same issue trying to use a 500Hz esc on our multicopter. I mainly tried every possible workaround without finding a solution. 
Do you guys know where the 400 Hz cap is enforced? Even removing the code pointed by feemi has no effect on the final rate.

FlavioTonelli, 2021-06-24:
My bad. I confirm the Pam rate is capped in the code showed by feemi.
I got crazy for two days just for a make clean. PX4iofirmware is not recompiling without cleaning the build folder.