PX4/PX4-Autopilot #13765: need help (pixhawk 2.1 orange cube)

kimjongyoung, 2019-12-20:
I have bought Pixhawk 2.1 orange cube from ProfiCNC and
I am trying to build and upload the firmware following steps at https://dev.px4.io/v1.9.0/en/setup/building_px4.html.

In MacOS environment,
after building firmware using the command;

make px4_fmu-v3_default

followed by command

make px4_fmu-v3_default upload

which gives me an error message;

Loaded firmware for board id: 9,0 size: 1707721 bytes (82.07%), waiting for the bootloader...

Found board id: 140,0 bootloader version: 5 on /dev/tty.usbmodem144101
WARNING: Firmware not suitable for this board (board_type=140 board_id=9)

My question is

1. Do I have to install the bootloader? and I want to know how (or where can I get the relevant information for the Orange cube)

2. What is the right make command for the Orange cube?

jzazbert, 2019-12-20:
The proficnc cubes seem to be coming with a new ardupilot bootloader which QGC and PX4 build tools does not recognize. When I messed around with an orange cube, I had to build the bootloader from the repo and flash it on.

Secondly, the orange cube is an H7 processor, with different sensors and such. It was not compatible with px4_fmu-v3, or any of the existing PX4 build targets, as of 1.9.2 when I was messing with it. There are some H7 bits in the code now which might be a good place to start.

atr2600, 2020-01-10:
@jzazbert have you had any more success working with the Orange cube with the H7 processor? 

I really want to start using PX4 and not ardupilot.

jzazbert, 2020-01-11:
I’ve unfortunately had other distractions and haven’t been able to work on it. Good luck!

JacobCrabill, 2020-01-23:
@atr2600 one thing you could try is adapting the Holybro Durandal-v1 build target to the Orange Cube.  The microcontrollers used within both are (nearly) identical, so it's a very close starting point.  As @jzazbert pointed out though, you'll have to first load the PX4 bootloader on the Cube to be able to load firmware.

JacobCrabill, 2020-05-19:
Orange Cube support is in progress: https://github.com/PX4/Firmware/pull/14928

hamishwillee, 2020-08-23:
PX4 now has Cube Orange support in master. I'm going to close this. Re-test and reopen if you still get this problem.

closed 2020-08-23 by hamishwillee

djy4, 2021-02-14:
Nope, still a problem. Downloaded the latest master and tried uploading with the same error.

In Linux environment,  after building firmware sucessfully 
make px4_fmu-v3_default upload

WARNING: Firmware not suitable for this board (Firmware board_type=140 board_id = 9)

JacobCrabill, 2021-02-14:
`fmu-v3` is the build target for the Black Cube (and related variants).  Use the Orange Cube build target (`cubepilot_cubeorange`)

djy4, 2021-02-14:
makes sense however the command:
make px4_cubepilot_cubeorange_default 

gives the following: "Make target px4_cubepilot_cubeorange not found."

I am just getting started with PX4 so I probably have missed a step somewhere...I just don't know where. the documentation seems pretty straight forward: 
http://docs.px4.io/master/en/dev_setup/building_px4.html

SEUZTh, 2021-03-07:
@JacobCrabill Hi, could you tell me detailed name? cubepilot_cuberoange