PX4/PX4-Autopilot #13788: Intermittent arming and flying issues in mission mode even with position lock

myeabhishek, 2019-12-25:
Hi team, I have been using PX4 since last 3 years. I have been facing intermittent issues with arming and flying in mission mode across various versions.

The exact use case is:

After uploding a valid mission on the drone, I try to start mission using qgroundcontrol. I have observed following intermittent issues with this:

1. Drone fails to arm with EKF internal checks errors. I adjusted COM_ARM parameters but still I face issues even with proper calibration of all sensors.
2. Drone gets armed but fails to switch to mission mode therefore automatically disarms after 5 seconds as defined in the params.
3. Drone armed, mode is switched to mission but the drone doesnt fly.

I face these issues only intermittently. Ideally it does fly. I am looking for more reliability wherein my drone flies each time or atleast gives meaningful errors which can be comprehended. 

I can look and fix the issues, I just need some pointers onto where the issue could lie. 

Thankyou in advance.

Hardware details:
Pixhawk 2
Here 2

Software details:
PX4 firmware: V1.7.2
Multicopter
Gcs: qgroundcontrol

dagar, 2019-12-29:
Have you tried reproducing these issues with newer PX4 (v1.10.0 or master)?

I'd start with gathering a log from boot that captures the system state when attempting to arm.

bresch, 2020-03-30:
There have been a ton of improvements since v1.7 and this issue isn't relevant anymore, please reopen if the issue is still present for you on the latest version.

closed 2020-03-30 by bresch