PX4/PX4-Autopilot #13793: Altitude Fluctuation in Constant-Altitude Missions

jordancoult, 2019-12-27:
**Describe the bug**
When flying a basic mission with constant-altitude waypoints, the drone’s altitude erroneously fluctuates. Our use-case requires a much more reliable altitude for precision deployment of sensors. Let me know if you can/can't reproduce this issue.

[image]

Notice how the estimated altitude doesn't follow the setpoint altitude in mission mode (purple region), but it follows exactly in position mode (green region). This shows that the instruments and sensors are capable of producing altitude-accurate flight.

I also posted this issue here: [https://discuss.px4.io/t/altitude-fluctuation-in-missions/14416](url)

**To Reproduce**
1. Use Pixhawk1 or Pixhawk Cube with 1.8.2 or 1.9.2 firmware
2. Upload mission detailed below
3. Take off and start mission
4. See drone fluctuate 1-2 meters from setpoint.

Mission:

Mission Start - 0m
Takeoff - 8m rel to home
Waypoint - 8m rel to home, 5sec hold
Waypoint - 8m rel to home, 5sec hold
Waypoint - 8m rel to home
Waypoint - 8m rel to home
Waypoint - 8m rel to home
Waypoint - 8m rel to home

**Expected behavior**
I expect the drone's altitude to follow the setpoint altitude in mission mode with the same accuracy as position mode.

**Log Files Analysis**
[https://review.px4.io/plot_app?log=624517a3-9159-415c-8974-4fc790206f6f](url)
More log files for different missions are available upon request. They have all resulted in the same issue--despite using different setpoint altitudes and types of waypoints.

**Drone**
This problem was reproduced on two drones; a Flamewheel 450 and a custom drone that is much larger--about 1 meter span between opposite motors. We tried multiple configurations for each drone consisting of Pixhawk 1, Pixhawk Cube, v1.8.2 PX4 and v1.9.2 PX4. For each Pixhawk, we used the standard recommended GPS. Both drones were adequately powered.

dagar, 2019-12-29:
Have you tried tuning the position controller? MPC_Z_P, MPC_Z_VEL_P, MPC_Z_VEL_I, etc.
I would also encourage you to to sync up with current PX4 (v1.10.0).

jordancoult, 2019-12-30:
@dagar thanks for the suggestions. I tried missions on v1.10.0 and played with the MPC_Z parameters. I didn't see any improvements. If it were an issue with those parameters, I would expect the altitude to also fluctuate in other auto-altitude modes, which doesn't happen.

bresch, 2022-08-15:
Probably some setup issues causing bad velocity and/or position estimates
[image]

closed 2022-08-15 by bresch