PX4/PX4-Autopilot #13799: Offboard Mode not Switching after Arming

lilcompaadres, 2019-12-28:
Issue: Switching to offboard after arming sets off error on real-world Drone, however switching straight into offboard is possible

Steps:
1. ROS_Node is run that:
       1.Arms drone
       2. Then sends stream of setpoints 
       3. Switches to offboard mode
2. Warning is displayed on PX4 log
      [ WARN] [1577568026.571076614]: CMD: Unexpected command 176, result 0
3. While the drone is still armed and receiving setpoints, switching to offboard via RC is possible but exits after a split second.

PX4 LOG:
NODES
  /
    mavros (mavros/mavros_node)

auto-starting new master
process[master]: started with pid [9871]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ae23efc6-29b7-11ea-8058-00044be6704f
WARNING: Package name "Drone" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "schoolThings" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosout-1]: started with pid [9882]
started core service [/rosout]
process[mavros-2]: started with pid [9890]
[ INFO] [1577567950.983533636]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1577567950.993841604]: serial0: device: /dev/ttyACM0 @ 57600 bps
[ INFO] [1577567950.996088252]: GCS bridge disabled
[ INFO] [1577567951.195053799]: Plugin 3dr_radio loaded
[ INFO] [1577567951.205287286]: Plugin 3dr_radio initialized
[ INFO] [1577567951.205849430]: Plugin actuator_control loaded
[ INFO] [1577567951.218821136]: Plugin actuator_control initialized
[ INFO] [1577567951.349121587]: Plugin adsb loaded
[ INFO] [1577567951.360487747]: Plugin adsb initialized
[ INFO] [1577567951.361059891]: Plugin altitude loaded
[ INFO] [1577567951.364989586]: Plugin altitude initialized
[ INFO] [1577567951.365421988]: Plugin cam_imu_sync loaded
[ INFO] [1577567951.367913849]: Plugin cam_imu_sync initialized
[ INFO] [1577567951.368404637]: Plugin command loaded
[ INFO] [1577567951.387956755]: Plugin command initialized
[ INFO] [1577567951.388408532]: Plugin companion_process_status loaded
[ INFO] [1577567951.393551110]: Plugin companion_process_status initialized
[ INFO] [1577567951.393931220]: Plugin debug_value loaded
[ INFO] [1577567951.405495612]: Plugin debug_value initialized
[ INFO] [1577567951.405627437]: Plugin distance_sensor blacklisted
[ INFO] [1577567951.406209217]: Plugin fake_gps loaded
[ INFO] [1577567951.434577196]: Plugin fake_gps initialized
[ INFO] [1577567951.435246633]: Plugin ftp loaded
[ INFO] [1577567951.457281027]: Plugin ftp initialized
[ INFO] [1577567951.457754368]: Plugin global_position loaded
[ INFO] [1577567951.496433856]: Plugin global_position initialized
[ INFO] [1577567951.496840737]: Plugin gps_rtk loaded
[ INFO] [1577567951.502291809]: Plugin gps_rtk initialized
[ INFO] [1577567951.502772858]: Plugin hil loaded
[ INFO] [1577567951.535181679]: Plugin hil initialized
[ INFO] [1577567951.535688874]: Plugin home_position loaded
[ INFO] [1577567951.545151777]: Plugin home_position initialized
[ INFO] [1577567951.545810537]: Plugin imu loaded
[ INFO] [1577567951.564496287]: Plugin imu initialized
[ INFO] [1577567951.565104525]: Plugin landing_target loaded
[ INFO] [1577567951.607008645]: Plugin landing_target initialized
[ INFO] [1577567951.607483444]: Plugin local_position loaded
[ INFO] [1577567951.622882426]: Plugin local_position initialized
[ INFO] [1577567951.623460039]: Plugin log_transfer loaded
[ INFO] [1577567951.633395970]: Plugin log_transfer initialized
[ INFO] [1577567951.633889258]: Plugin manual_control loaded
[ INFO] [1577567951.640757330]: Plugin manual_control initialized
[ INFO] [1577567951.641162076]: Plugin mocap_pose_estimate loaded
[ INFO] [1577567951.649337303]: Plugin mocap_pose_estimate initialized
[... 84 lines cut ...]
[ INFO] [1577568018.183989078]: FCU: ARMED by Arm/Disarm component command
[ INFO] [1577568018.230233264]: FCU: [logger] file: /fs/microsd/log/2019-12-28/21_20_1
[ INFO] [1577568022.202710866]: HP: requesting home position
[ WARN] [1577568026.571076614]: CMD: Unexpected command 176, result 0
[ INFO] [1577568028.180467556]: FCU: DISARMED by Auto disarm initiated
[ INFO] [1577568032.202880597]: HP: requesting home position
[ INFO] [1577568042.203011109]: HP: requesting home position
^C[mavros-2] killing on exit
Debug:   mavconn: serial0: recv: v2.0 !CRC Message-Id: 111 [13 bytes] IDs: 1.1 Seq: 147
Debug:   mavconn: serial0: recv: v2.0 !CRC Message-Id: 65 [42 bytes] IDs: 1.1 Seq: 148
Debug:   mavconn: serial0: recv: v2.0 !CRC Message-Id: 141 [32 bytes] IDs: 1.1 Seq: 149
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Drone Description:
-QuadCopter
-PX4 
-Jetson Nano companion
**Additional context**
Add any other context about the problem here.

**Code used for the ROS_NODE works in Gazebo simulation

julianoes, 2020-01-30:
Thanks for the report. I have seen similar complaints. Have you found a solution or workaround?
I still need to understand what is going on.

DiegoHerrera1890, 2021-02-03:
Hello! I am facing the same issue. Is there an answer?

hsilvaga, 2021-02-04:
Not a solution but more of a workaround... I reverted the pixhawk firmware to version 1.10.0 and that seemed to fixed my issue.

Jaeyoung-Lim, 2021-02-04:
@hsilvaga Looking at your log outputs such as
```
ERROR] [1577567952.856373856]: FCU: REJECT POSITION CONTROL
[ WARN] [1577567986.446954895]: GP: No GPS fix
```
You don't have a valid position estimate. Therefore you cannot use position/velocity setpoints

ninja-hu, 2021-07-16:
I saw a similar error for the latest stable build v1.12.0. Manual flight seems to work fine, but the offboard command is rejected when using mavros.

I then reverted to v1.11.0 and it worked. Haven't tried other versions yet.

julianoes, 2021-08-30:
I'm going to close this. Please comment again if this is still an issue.

closed 2021-08-30 by julianoes