PX4/PX4-Autopilot #13809: Build of px4_fmu-v4_default fails on Fedora 31 with GCC 9.2.1 due to -Werror=address-of-packed-member

stou-sandalski, 2019-12-29:
**Describe the bug**

Building a fresh copy of the master fails with: 

../../src/lib/parameters/flashparams/flashfs.c:190:2: error: converting a packed 'flash_entry_header_t' {aka 'struct flash_entry_header_t'} pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]

**To Reproduce**
Steps to reproduce the behavior:
Following build instructions on https://dev.px4.io/v1.9.0/en/setup/building_px4.html

1.  git clone https://github.com/PX4/Firmware.git
2. cd Firmware
3. make px4_fmu-v4_default

**Expected behavior**

A working build

**Full output**

[stou@cipher Firmware]$ make px4_fmu-v4_default
-- PX4 version: v1.11.0-beta1-121-gb3889e6187
-- PX4 config file: /tmp/px4/Firmware/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.2.0
-- The C compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib64/ccache/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/lib64/ccache/arm-none-eabi-gcc
-- Check for working C compiler: /usr/lib64/ccache/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "3.7.5") 
-- Found PY_jinja2: /usr/lib/python3.7/site-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_extras
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/px4/Firmware/build/px4_fmu-v4_default
[0/1277] git submodule platforms/nuttx/NuttX/nuttx
[2/1277] git submodule platforms/nuttx/NuttX/apps
[7/1277] git submodule src/lib/ecl
[9/1277] git submodule src/drivers/gps/devices
[13/1277] git submodule src/drivers/uavcan/libuavcan
[17/1277] git submodule mavlink/include/mavlink/v2.0
[512/1277] Building C object src/lib/parameters/flashparams/CMakeFiles/flashparams.dir/flashfs.c.obj
FAILED: src/lib/parameters/flashparams/CMakeFiles/flashparams.dir/flashfs.c.obj 
/usr/bin/ccache /usr/lib64/ccache/arm-none-eabi-gcc -DCONFIG_ARCH_BOARD_PX4_FMU_V4 -DMODULE_NAME=\"flashparams\" -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I../../boards/px4/fmu-v4/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/stm/stm32f4/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fno-common -Wbad-function-cast -Wnested-externs -Wstrict-prototypes -Wno-sign-compare -Wno-cast-align -std=gnu11 -MD -MT src/lib/parameters/flashparams/CMakeFiles/flashparams.dir/flashfs.c.obj -MF src/lib/parameters/flashparams/CMakeFiles/flashparams.dir/flashfs.c.obj.d -o src/lib/parameters/flashparams/CMakeFiles/flashparams.dir/flashfs.c.obj   -c ../../src/lib/parameters/flashparams/flashfs.c
../../src/lib/parameters/flashparams/flashfs.c: In function 'blank_check':
../../src/lib/parameters/flashparams/flashfs.c:190:2: error: converting a packed 'flash_entry_header_t' {aka 'struct flash_entry_header_t'} pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  190 |  uint32_t *pm = (uint32_t *) pf;
      |  ^~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
[518/1277] Building CXX object src/drivers/uavcan/uavcan_drivers/stm32/driver/CMakeFiles/uavcan_stm32_driver.dir/src/uc_stm32_thread.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:195: px4_fmu-v4_default] Error 1

leekyubum, 2020-01-30:
how to solve this problem?

stou-sandalski, 2020-01-30:
I think these errors are caused by the newer gcc being more pedantic and treating some warnings as errors. It's possible to disable these warning/errors by placing GCC diagnostic pragmas at the top of the 
"broken" files. 

To get it to build I had to use one of the following pragmas at the top of a few files (flashfs.c, IridiumSBD.cpp, px4io.cpp, test_file.c, test_mount.c, test_time.c):

#pragma GCC diagnostic ignored "-Waddress-of-packed-member"
#pragma GCC diagnostic ignored "-Wcast-function-type"
#pragma GCC diagnostic ignored "-Wvla-larger-than="
#pragma GCC diagnostic ignored "-Wabsolute-value"

This is a **hack** and probably good to assume that it's not entirely safe to fly with the resulting build.

julianoes, 2020-01-31:
Note that we currently only support GCC 7-2017-q4 as described in https://dev.px4.io/v1.9.0/en/setup/dev_env_linux_centos.html#gcc-toolchain-installation.

If you want to use a newer compiler, I suggest that you fix the warnings, or remove the `-Werror` flag, so that warnings do not fail the build.
https://github.com/PX4/Firmware/blob/b14e5568d3aacf77f04d222ed8c2d67ebce08309/cmake/px4_add_common_flags.cmake#L71

closed 2020-01-31 by julianoes

zhoushanjinjie, 2023-02-03:
> **Describe the bug**
>
> Building a fresh copy of the master fails with:
>
> ../../src/lib/parameters/flashparams/flashfs.c:190:2: error: converting a packed 'flash_entry_header_t' {aka 'struct [...]

Hi, did you solve the your GCC problem? I am currently also switched to Fedora 37