PX4/PX4-Autopilot #13817: Standard VTOL mismatch model in SITL with visualization

Jaeyoung-Lim, 2019-12-30:
**To Reproduce**
You can run the stadard_vtol model with the following command
```
make px4_sitl gazebo_standard_vtol
```
The simulation works as expected and flies nicely. The visualization shows a flying wing model with rotors attached. However, the standard_vtol model has a rudder and elevator and the mixer is using the elevator and elevons don't provide pitch commands: https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/mixers-sitl/standard_vtol_sitl.main.mix 

[image]

Would it be more desirable to make a flying wing model with rotors for hovering? or switch the mesh to a more appropriate model (maybe a cessna with rotors)?

What was the intention of having the elevator and rudder?

**Expected behavior**

The model flies as expected in the simulation

Jaeyoung-Lim, 2019-12-30:
This seems to be due to a limit coming from the aerodynamics modeled in gazebo

closed 2019-12-30 by Jaeyoung-Lim

frontw, 2020-01-15:
Hi! What a limit and how it was resolved?
Is there a Gazebo model of VTOL with ailerons and tail rudder and elevator available?

frontw, 2020-01-22:
Here is standard VTOL visually correct model https://www.youtube.com/watch?v=Ifn4k6Rtkrc.
Where to find it? Why it is not used?