PX4/PX4-Autopilot #13829: can't view gazebo video in qground control

liangpu597, 2020-01-02:
I want to view gazebo video in qgc.
I follow the instance in: https://dev.px4.io/v1.9.0/en/simulation/gazebo.html
My Step is :

Enable the GStreamer Plugin (if disabled) by changing the BUILD_GSTREAMER_PLUGIN option to "ON" in /Tools/sitl_gazebo/CMakeLists.txt (as shown below):
option(BUILD_GSTREAMER_PLUGIN "enable gstreamer plugin" "ON")

make clean
make px4_sitl gazebo_typhoon_h480

I open qgc and in settings-> general -> video source
and its no video available. I can't find other choice.
my gazebo version is : 9.10
QGC version is : v3.5.4

liangpu597, 2020-01-02:
I have solved , thanks

closed 2020-01-02 by liangpu597

ChenXL765, 2020-04-01:
hi, I think I had the same problem as you, so i want to know how did you solve it. Thank you very much!