PX4/PX4-Autopilot #13856: Intel Aero build configuration problem

altshuler, 2020-01-05:
**Describe the bug**
While building Intel Aero default configuration on latest master
strange behavior observed. Created build/intel_aerofc-v1_default
folder, but inside modalai_fc-v1_default.px4     modalai_fc-v1_default.elf    and     
modalai_fc-v1_default.map   binary files  generated.
Same behavior seen on v1.10.0.

**To Reproduce**
Steps to reproduce the behavior:
issue     **make intel_aerofc-v1_default**    command

**Expected behavior**
Correct binaries generation for intel_aerofc-v1_default configuration

**Log Files and Screenshots**
evgeny@quad:~/Firmware$ make intel_aerofc-v1_default
-- PX4 version: v1.11.0-beta1-144-g5dfc8f1
-- PX4 config file: /home/evgeny/Firmware/boards/modalai/fc-v1/default.cmake
-- PX4 config: modalai_fc-v1_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/evgeny/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Enabling double FP precision hardware instructions
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/evgeny/Firmware/build/intel_aerofc-v1_default
[1/1236] git submodule platforms/nuttx/NuttX/nuttx
[11/1236] git submodule platforms/nuttx/NuttX/apps
[12/1236] git submodule src/lib/ecl
[15/1236] git submodule src/drivers/gps/devices
[16/1236] git submodule src/drivers/uavcan/libuavcan
[18/1236] git submodule mavlink/include/mavlink/v2.0
[1234/1236] Linking CXX executable modalai_fc-v1_default.elf
Memory region         Used Size  Region Size  %age Used
            itcm:          0 GB      2016 KB      0.00%
           flash:     1614593 B      2016 KB     78.21%
            dtcm:          0 GB       128 KB      0.00%
           sram1:       47304 B       368 KB     12.55%
           sram2:          0 GB        16 KB      0.00%
[1236/1236] Creating /home/evgeny/Firm...c-v1_default/modalai_fc-v1_default.px4

**Drone (please complete the following information):**
INTEL AERO RTF

dagar, 2020-01-06:
I'm not seeing this locally or on the build system. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile/detail/master/1902/pipeline/70

[image]

What happens if you try to build another target like `px4_fmu-v5_default`?

altshuler, 2020-01-06:
@dagar , px4_fmu-v5_defaul  built properly. also tried another targets and 
everything is fine.

dagar, 2020-01-06:
This is reproducible if you make clean?

``` Console
make clean
make intel_aerofc-v1_default
```

altshuler, 2020-01-06:
Yes. Also Removed and downloaded the whole Firmware repository.
Also synchronized subprojects with   git submodule update --init --recursive

altshuler, 2020-01-09:
@dagar , any updates on this ?

I tried to compile intel_aerofc-v1_default  target on the drone.
I have ubuntu 16.04 installed on it. The resuls is the same...

Also I saw that the last  tag that compiled properly was v1.10.0-beta3.
On v1.10.0-beta4  was added ModalAI target configuration.
On this version I have modalai_fc-v1_default  binaries after compilation  inside 
build/intel_aerofc-v1_default  folder.
Maybe it some how related ?

altshuler, 2020-01-14:
I think I found the problem.
In this file:
https://github.com/PX4/Firmware/blob/4f362f583511b6326fda859934905e553f74eb03/cmake/px4_config.cmake#L90
in targets    intel_aerofc-v1_default    and     modalai_fc-v1_default  
${model}_${label}    fields are the same -  **fc-v1**    and its match when make goes
through the list of targets. When I change this line and add **vendor** field:
if ((${config_no_os} MATCHES "${vendor}_${model}_${label}"))

correct binary files generated in Firmware/build/intel_aerofc-v1_default folder.
I'm Still have a problem with flashing it...
don't know why this problem is not appears in automatic build.
@dagar , can you look at this please ?

dagar, 2020-02-08:
Fixed in https://github.com/PX4/Firmware/pull/14123.

closed 2020-02-08 by dagar