PX4/PX4-Autopilot #13857: Inquiry and discovery of Fault Tolerant control system in PX4 in case of Actuator Faults mid flight.

ucalyptus, 2020-01-05:
I was reading this paper and so I am here to know the scope of this technique in PX4 firmware. I want to know if this can be implemented in PX4 and what should be the overall approach and strategy. 
Paper : **Actuator Fault Detection and Fault-Tolerant Controlfor Hexacopter** [LINK](http://www.mdpi.com/1424-8220/19/21/4721/)

Some tips would suffice on the overall implementation and design strategy.

Also please note this/similar question was unanswered on the discuss forum [link here.](https://discuss.px4.io/t/motor-fault-tolerance-for-multicopter/7746)

Looking forward to some guidance and insights into the scope of this paper into becoming an actual working implementation in PX4.
cc: @LorenzMeier 

Thanks,
Sayantan

julianoes, 2020-02-14:
Maybe @jlecoeur you want to comment here.

jlecoeur, 2020-02-14:
Hi @ucalyptus 
I started working on dynamic mixing to address the fault tolerance part (not the fault detection). You can find it here :
https://github.com/PX4/Firmware/pull/13351
The approach is to dynamically recompute the mixer in case of motor failure.
Here is a demo with an hexacopter loosing one then two motors:
https://www.youtube.com/watch?v=yR5okaHITpg
Feel free to help if you will, that would be much appreciated!

That said, is all you want is a fault tolerant hexacopter, there would be easier ways. Because the work linked above aims to be applicable for all kinds of robots, it involves computing pseudoinverse on-board, which is pretty involved.
For a more limited case, say an hexacopter in x config, a simpler alternative would be to precompute the mixers for all failure cases (6 times one motor failure cases + 2-among-6 double motor failure cases) at compile time and avoid the pseudoinverse.

closed 2020-05-15 by julianoes