PX4/PX4-Autopilot #13868: SITL issues

RomanBapst, 2020-01-06:
Throwing in some things I noticed when testing in SITL with jmavsim on lastest master.

- [ ] console hangs when you do a "sensors stop"

- [ ]  If you stop the simulator from publishing mag data and then try to takeoff you get a PreflightFail indicating that there is no valid mag data. But the system does not stop you from taking off, is this intentionally?

Furthermore, I tried to debug first issue using gdb but had not luck getting something to work.

[image]

julianoes, 2020-01-06:
Can you create two separate issues please? This is hard to manage like that.

closed 2020-01-06 by julianoes

dagar, 2020-01-06:
>  If you stop the simulator from publishing mag data and then try to takeoff you get a PreflightFail indicating that there is no valid mag data. But the system does not stop you from taking off, is this intentionally?

@RomanBapst definitely not intentional, and could be more than a SITL bug.

RomanBapst, 2020-01-06:
@dagar Ok, will investigate further

RomanBapst, 2020-01-07:
@dagar Ok, apparently it does it because the arming state is set to IN_AIR_RESTART when you do a commander takeoff. It works properly when you send a takeoff command from QGC.
Not sure if that is intentional again :-D