PX4/PX4-Autopilot #13875: Dual rotor VTOL simulation

YuCodes, 2020-01-07:
As for the present release v1.10.0, tailsitter simulation which is gazebo_tailsitter is based on 4 rotor tailsitter. If this could be extended to a dual rotor platform with elevons, it could be useful for many. 

I came across an opensource project that simulates this using PX4 called PX4-PhoenixDrone which was conducted in Toronto. I think it could be integrated into the main stack as it already has a model and the sdf file for dual rotor tailsitter configuration.

https://github.com/utiasSTARS/PhoenixDrone-sitl_gazebo/tree/ed3b70cbd7612746db90d3d091c1ba3767f4f151/models/phoenixdrone

Jaeyoung-Lim, 2020-01-07:
@YuCodes I looked into the code, but the dynamics in the repo you mentioned is a approximate model of the tailsitter in hover, therefore it doesn't work in fixed wing mode

YuCodes, 2020-01-07:
@Jaeyoung-Lim Thank you much for the update, I went through the thread. Do you think that is there any-way to model the prop wash in the gazebo, so that dual-rotor configuration is somehow possible to simulate? Any pointers?

Jaeyoung-Lim, 2020-01-07:
@YuCodes I think I do have some ideas, but I can't promise you any delivery dates

Is there a special reason why you need the duo tailsitter? Given the abstraction px4 mixer does, I think most cases you can test with the quad tailsitters no?

YuCodes, 2020-01-08:
yeah, I'm currently testing the platform with the quad tailsitter. I intend to flight test a platform, that i designed. I was thinking to get an understanding of flight characteristics especially the transition before I move on to the actual test flight.
[image]

Jaeyoung-Lim, 2020-01-09:
@YuCodes Gazebo SITL simulations are not intended to accurately show the flight characteristics of a vehicle. This is true for all multirotor and fixed wing models too.

It is more designed to check the sanity of the software, so the dynamics is "good enough" to check the software functionalities. I would like to support dual rotor tail sitters just to check the mixer functioning properly, but this will always be far from reality given the priorities we have. (e.g. execution speed over accuracy)

YuCodes, 2020-01-10:
@Jaeyoung-Lim thanks for the explanation, I understand your point on not being accurate about flight characteristics. what I actually needed to test is more what you described, whether the mixers are configured right, so that in transition, it behaves as it should behave.

lum-supakorn, 2020-06-09:
Any news on this? I'm developing a model for a complete dual rotor flight envelope too.

Jaeyoung-Lim, 2020-06-09:
@lum-supakorn You can test all of the firmware functions with the current tailsitter model given the mixer architecture of px4. The only difference would be the control authority difference at low speed which will be platform specific.

lum-supakorn, 2020-06-12:
Thank you! Do you know why the current tailsitter model needs a virtual elevator? Can it not control pitch using just the elevons?

Jaeyoung-Lim, 2020-06-12:
@lum-supakorn That comes from the liftdrag plugin not being able to properly model the moments generated by the lift.

lum-supakorn, 2020-06-12:
> @lum-supakorn That comes from the liftdrag plugin not being able to properly model the moments generated by the lift.

If the virtual elevator generate pitching moment by generating lift off-centered, can't the elevons do the same and control the pitch as well? Why do we need a separate pitch controller?

Jaeyoung-Lim, 2020-06-12:
@lum-supakorn Those two are physically different phenomenons. The moments that are generated by elevons are induced by the wing itself and not by generating lift off the CG. 

The reason that the moments are not being modeled in the liftdrag plugin can be found here: https://github.com/PX4/sitl_gazebo/blob/master/src/liftdrag_plugin/liftdrag_plugin.cpp#L372

lum-supakorn, 2020-06-12:
> @lum-supakorn Those two are physically different phenomenons. The moments that are generated by elevons are induced by the wing itself and not by generating lift by an offset from the CG.
> 
> The reason that the moments are not being modeled in the liftdrag plugin can be found here: https://github.com/PX4/sitl_gazebo/blob/master/src/liftdrag_plugin/liftdrag_plugin.cpp#L372
> 
> Since the moment of all control surfaces are set to zero, the elevator is generating lift and therefore generating moment around the vehicle.

I'm quite confused. Aren't the elevons responsible for increasing/decreasing the left/right wing CL and (because the lift of each wing is applied away from CG) therefore controlling the rolling moment of the vehicle?

Jaeyoung-Lim, 2020-06-12:
@lum-supakorn No, that is not how the pitch of flying wings are controlled. It is affected by both the moment generated from the wing and the moment generated from the lift of the wing.

lum-supakorn, 2020-06-12:
> @lum-supakorn No, that is not how the pitch of flying wings are controlled. It is affected by both the moment generated from the wing and the moment generated from the lift of the wing. Please revisit how the dynamics actually work for flying wings.

Thank you!

lum-supakorn, 2020-06-16:
Proven. I have hard-coded pitching moment control using data obtained from XFLR5 and the tailsitter SITL model is now able to fly without the virtual elevator. Working on properly reading moment coefficient data.

Jaeyoung-Lim, 2020-06-16:
@lum-supakorn Could you maybe make a PR on fixing the [delta-wing model](https://github.com/PX4/sitl_gazebo/blob/master/models/delta_wing/delta_wing.sdf) in [PX4/sitl_gazebo](docker.io/auterion/simulation-ubuntu:build-v1.4.0.5-dev1)?

lum-supakorn, 2020-06-16:
> @lum-supakorn Could you maybe propagate this and make a PR on fixing the [delta-wing model](https://github.com/PX4/sitl_gazebo/blob/master/models/delta_wing/delta_wing.sdf) in [PX4/sitl_gazebo](https://github.com/PX4/sitl_gazebo)?

I would love to but I’m trying to figure out which approach I should implement regarding reading CL, CD, and CM data.

Could you please take a look at https://github.com/PX4/sitl_gazebo/issues/519

hvourtsis, 2021-07-07:
Hello! Are there any updates on this topic? Thanks a lot!

Jaeyoung-Lim, 2021-07-07:
@hvourtsis This is currently limited with the aerodynamics modeling capabilities of the liftdrag_plugin in the post stall regime.

hvourtsis, 2021-07-07:
Thank you for the prompt reply @Jaeyoung-Lim. Does this also mean that prop-wash is still not modeled? 
I saw some ongoing work on the plugin and also that you mentioned in another topic that there would be work on a new [https://github.com/PX4/PX4-SITL_gazebo/issues/684#issuecomment-752801575](https://github.com/PX4/PX4-SITL_gazebo/issues/684#issuecomment-752801575).

lum-supakorn, 2021-07-07:
@hvourtsis For slipstream aerodynamics take a look at [PhoenixDrone](https://github.com/utiasSTARS/PhoenixDrone). Slipstream forces and moments are empirically modeled as functions of motor angular velocity and elevon deflection angle.

harshilairbound, 2022-12-29:
Hello!
I'm trying to simulate a tailsitter as well. Reading the above discussion, it seems that using gazebo to test out the more nuanced stuff, like the effects of wingtips and wing shapes on the flight performance might not be a good option. 
Does anyone have any suggestions on how to simulate flights with that fidelity?

lum-supakorn, 2022-12-29:
Hi, @harshilairbound. I'd suggest you employ computational or experimental aerodynamics methods to study the effects of wingtips and wing shapes on your vehicle, then import the resulting data into Gazebo dynamics simulation. What is going on in the PX4-Gazebo simulation stack is that the aerodynamic loads (forces and moments) on the vehicle are already prescribed somewhere in the model. For example, these loads might be prescribed as functions of control surface deflection angle and the rotational speed of the propeller. Gazebo then uses these loads to calculate the dynamics (linear and angular velocity, etc.) of the vehicle being simulated. The physics of fluid and the vehicle is not directly simulated in Gazebo since it is only a dynamics simulator, not a computational fluid dynamics package.

Cheers

HarshilNaikk, 2023-01-18:
Hey @lum-supakorn 
Thank you for your reply. I am doing just that for now! We are simulating different vehicle geometries in a CFD software, and getting lift and drag coefficient values, which I am looking to put directly into the advanced lift drag plugin. Unfortunately, I am a little new to this and am still figuring out how to setup the plugin. @Jaeyoung-Lim any ideas or pointers? 
Thank you so much in advance.

Best

HarshilNaikk, 2023-01-19:
From what I have figured out yet, the libLiftDragPlugin.so file being used in the sdf is generated during the build process. I wish to know how to use the Advanced Lift Drag Plugin in this way. 
Thanks again!

Jaeyoung-Lim, 2023-01-19:
@HarshilNaikk It is only using jinja templates to configure the UDP/TCP ports and nothing else. You can also do so for the advanced liftdrag plugin

harshilairbound, 2023-01-26:
Right. I was able to successfully use it. Thanks a lot!

The issue I'm currently facing is, my duo tailsitter doesnt takeoff in a stable way in gazebo. It always crashes, and has no roll stability apparently. I think it has something to do with the differential thrust not being part of the mixer, since I made this model by repurposing the existing quad tailsitter model. I also went through the Control Allocation code, and cannot figure out what's wrong. 

Any ideas on this? 
@lum-supakorn did you also face this? 

Thanks a ton again!

mengchaoheng, 2024-12-04:
> Do you think that is there any-way to model the prop wash in the gazebo, so that dual-rotor configuration is somehow possible to simulate? Any pointers?

Yes, it is actually only necessary to modify the src/liftdrag_plugin/liftdrag_plugin.cpp file in sitl_gazebo so that the wind on the control surface is not only generated by the forward flight speed, but also by the rotation of the propeller, so that the pitch moment can be generated in vertical mode. 

> Does this also mean that prop-wash is still not modeled?

@hvourtsis  Although it has been 4 years since the author asked the question, I found that no one in the px4 development team seems to have improved this, because the latest version (v1.15.x) of the Tailsitter model has not changed, and four motors are still used instead of control surfaces to generate roll/pitch torque in vertical flight mode. Maybe the development team is committed to using a newer gazebo simulation instead of gazebo-classic. 

> Unfortunately, I am a little new to this and am still figuring out how to setup the plugin.

@lum-supakorn @HarshilNaikk  To simulate your dynamic model in the gazebo plug-in, first look at the design logic of the gazebo plug-in. A simple aerodynamic model such as [aerodynamics](https://classic.gazebosim.org/tutorials?tut=aerodynamics).

> Does anyone have any suggestions on how to simulate flights with that fidelity?

@harshilairbound 
I have done some work in this regard. We designed a new type of aircraft DFUAV, similar to Musk's space X. I wrote a new mix file (after v1.14, QGC is used to configure the actuator, and there is no need to write the mixing control file yourself). I conducted flight experiments with the help of pixhawk running px4, and also wrote a gazebo model of this new UAV, which was successfully run in px4 sitl. Refer to the [DFUAV](https://github.com/mengchaoheng/DuctedFanUAV-Autopilot) project, in which [liftdrag_plugin](https://github.com/mengchaoheng/DF_gazebo/blob/8c442e77b6523d6a6120f8d76413e712677155f7/src/liftdrag_plugin/liftdrag_plugin.cpp) is consider propeller. Another project  [PhoenixDrone](https://github.com/utiasSTARS/PhoenixDrone) that modifies liftdrag_plugin is [gazebo_ts_aerodynamic_plugin.cpp](https://github.com/utiasSTARS/PhoenixDrone-sitl_gazebo/blob/ed3b70cbd7612746db90d3d091c1ba3767f4f151/src/gazebo_ts_aerodynamic_plugin.cpp)

[image]

mengchaoheng, 2024-12-04:
> Do you think that is there any-way to model the prop wash in the gazebo, so that dual-rotor configuration is somehow possible to simulate? Any pointers? 

 Yes, it is actually only necessary to modify the src/liftdrag_plugin/liftdrag_plugin.cpp file in sitl_gazebo so that the wind on the control surface is not only generated by the forward flight speed, but also by the rotation of the propeller, so that the pitch moment can be generated in vertical flight mode. 

> Do you know why the current tailsitter model needs a virtual elevator? Can it not control pitch using just the elevons?

@lum-supakorn  As mention above.

> Does this also mean that prop-wash is still not modeled?

@hvourtsis 
Although it has been 4 years since the author asked the question, I found that no one in the px4 development team seems to have improved this, because the latest version (v1.15.x) of the Tailsitter model has not changed, and four motors are still used instead of control surfaces to generate roll/pitch torque in vertical flight mode. Maybe the development team is committed to using a newer gazebo simulation instead of gazebo-classic.

>Does anyone have any suggestions on how to simulate flights with that fidelity?

@harshilairbound  To perform dynamic modeling, you must first understand the design logic of gazebo plug-ins. A good example is the official document about [aerodynamics](https://classic.gazebosim.org/tutorials?tut=aerodynamics).

> Unfortunately, I am a little new to this and am still figuring out how to setup the plugin.

@HarshilNaikk I have done some work in this regard. We designed a new type of aircraft DFUAV, similar to Musk's SpaceX Starship. I wrote a new mix file (after v1.14, QGC is used to configure the actuator, and there is no need to write the mixing control file yourself). I conducted flight experiments with the help of pixhawk running px4, and also wrote a gazebo model of this new UAV, which was successfully run in px4 sitl. Refer to the [DFUAV](https://github.com/mengchaoheng/DuctedFanUAV-Autopilot) project, in which [liftdrag_plugin](https://github.com/mengchaoheng/DF_gazebo/blob/8c442e77b6523d6a6120f8d76413e712677155f7/src/liftdrag_plugin/liftdrag_plugin.cpp) is consider propeller. Another project  [PhoenixDrone](https://github.com/utiasSTARS/PhoenixDrone) that modifies liftdrag_plugin is [gazebo_ts_aerodynamic_plugin.cpp](https://github.com/utiasSTARS/PhoenixDrone-sitl_gazebo/blob/ed3b70cbd7612746db90d3d091c1ba3767f4f151/src/gazebo_ts_aerodynamic_plugin.cpp)

[image]