PX4/PX4-Autopilot #13881: building micrortps_agent failed

ldg810, 2020-01-08:
Hi, I am trying to using PX4 with ROS2 system.

Following the manual about the RTPS/ROS2 Interface, I successfully start px4_sitl_rtps and micrortps_client on the px4 in gazebo.

In case of micrortps_agent, using px4_ros_com(https://github.com/PX4/px4_ros_com) is almost fine but sensor_combined_listener is not working(No messages are received).

I am trying to use micrortps_agent in build directory inside of Firmware(Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent) but build is not sccessful with following error. (The foremost part of the error)

```
In file included from /home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:38:0,
                 from /home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:34:
/home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/home_position_Publisher.h:60:18: error: 'px4' has not been declared
     void publish(px4::msg::home_position* st);
```

I found that the difference between the px4_ros_com and Firmware is in header file.

For example,
sensor_combinedPubSubTypes.h in Firmware, there are no namespace such as px4 or px4_msgs
SensorCombinedPubSubTypes.h in px4_ros_com, there are px4_msgs namespace.

I think in Firmware code, there are some errors while generating mircortps_agent.

Anyone can help me? Thank you.

TSC21, 2020-01-08:
@ldg810 what Fast-RTPS version are you using?

ldg810, 2020-01-09:
> @ldg810 what Fast-RTPS version are you using? The recommended version for none-ROS2 usage is 1.8.2. But for ROS2, you have to use Fast-RTPS 1.7.2 with ROS2 Crystal, as we still don't support Dashing or Eloquent (though we are wroking on it).

Thank you for your answer.
My environment was ROS2 dashing and Fast-RTPS 1.8.2.
As you said, I changed versions of Fast-RTPS and ROS2 but same problem is occurred(px4 namespace problem).
I re-try with two Firmware versions(master branch, v1.10.0 tag) and ROS2 crystal(in Ubuntu Bionic) with Fast-RTPS 1.7.2.

However, the ROS2 crystal rosidl_adapter package make new compile error of encoding problem(I found that there are difference between rosidl_adapter and rosidl_parser for crystal and dashing version). So, I cannot build px4_msgs in ROS2 crystal version.

TSC21, 2020-01-09:
> However, the ROS2 crystal rosidl_adapter package make new compile error of encoding problem(I found that there are difference between rosidl_adapter and rosidl_parser for crystal and dashing version). So, I cannot build px4_msgs in ROS2 crystal version.

I am mostly sure you are having conflicts between versions. Did you remove Fast-RTPS 1.8.2 and ROS 2 Dashing from your system? Or have you made sure you are sourcing the right environemnt `setup.bash` from ROS 2? Just as a prove that it works, here's the latest and clean successful build on Travis CI: https://travis-ci.com/PX4/px4_ros_com.

TSC21, 2020-01-09:
Also, how exactly are you building the workspace? I recommend you using the scripts we have: https://github.com/PX4/px4_ros_com/blob/master/scripts/build_ros2_workspace.bash

ldg810, 2020-01-10:
> I am mostly sure you are having conflicts between versions. Did you remove Fast-RTPS 1.8.2 and ROS 2 Dashing from your system? Or have you made sure you are sourcing the right environemnt [...]

I continuously try to using Fast-RTPS for PX4 communication.
I decided to use ROS2 dashing(becuase it is longest LTS version at this time) and try another approach.
- use micrortps_agent with non-ROS2 version (remove env variables when making px4_sitl_rtps and use micrortps_agent in build directory)
- After the message reach to listener(non-ROS2 version), convert it to ROS2 topic

I succeeded to build non-ROS2 micrortps_agent and make listener for sensor_combined data and receive data from micrortps_client(px4_sitl_rtps in gazebo) -> micrortps_agent -> sensor_combined_listener
Now, I am working to use the data on the ROS2 system. when I list topics and node in this condition, I got this information.

```
root@ldg810-X299-WU8:/home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_listener# ros2 topic list
/parameter_events
/rosout
root@ldg810-X299-WU8:/home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_listener# ros2 node list
/trajectory_waypoint_subscriber
/position_setpoint_subscriber
/camera_capture_subscriber
/collision_report_subscriber
/obstacle_distance_subscriber
/debug_array_subscriber
/debug_key_value_subscriber
/debug_vect_subscriber
/position_setpoint_triplet_subscriber
/debug_value_subscriber
/camera_trigger_subscriber
/vehicle_trajectory_waypoint_subscriber
/optical_flow_subscriber
/vehicle_mocap_odometry_subscriber
/vehicle_visual_odometry_subscriber
/home_position_publisher
/wind_estimate_publisher
/radio_status_publisher
/satellite_info_publisher
/adc_report_publisher
/iridiumsbd_status_publisher
/vtol_vehicle_status_publisher
/airspeed_publisher
/sensor_selection_publisher
/cpuload_publisher
/sensor_baro_publisher
/battery_status_publisher
/estimator_status_publisher
/distance_sensor_publisher
/vehicle_odometry_publisher
/vehicle_attitude_publisher
/collision_constraints_publisher
/sensor_combined_publisher

root@ldg810-X299-WU8:/home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_listener# ros2 node info /sensor_combined_publisher
/sensor_combined_publisher
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: sensor_combined_publisher
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.6/src/rmw_node_info_and_types.cpp:87

root@ldg810-X299-WU8:/home/user/git/PX4_rtps/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_listener#
```

How do you think about this approach?

TSC21, 2020-01-10:
> I continuously try to using Fast-RTPS for PX4 communication.
> I decided to use ROS2 dashing(becuase it is longest LTS version at this time) and try another approach.
>
> * use micrortps_agent with [...]

That happens because the `micrrortps_agent` doesn't have a ROS2 compatible GUID, since it's not a ROS2 node. That can be eventually fixed but it's not on my priority list atm. If you can find a solution for that, please issue a PR and I am happy to review and merge :)

TSC21, 2020-01-10:
As a quick search, I could find that one can retrieve the RTPSParticipant GUID using https://github.com/eProsima/Fast-RTPS/blob/master/include/fastdds/rtps/participant/RTPSParticipant.h#L81.

TSC21, 2020-01-10:
The GUID is supposedly added to the Participant on its creation: https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/rtps/RTPSDomain.cpp#L118-L157.

So it might be that the DDS contexts are different and that's why its failing. But again, I don't expect this to work since the `micrortps_agent` by itself is not a ROS2 node: it's just a bridging node. It could be implemented eventually as a ROS2 node, but I don't have plans to do that now.

TSC21, 2020-01-10:
I am closing this issue as the original problem was answered. Please feel free to bring further discussions to the #messaging channel of the PX4 Slack.

closed 2020-01-10 by TSC21