PX4/PX4-Autopilot #13882: Duplicate UUID on real FC board

cjffly, 2020-01-08:
**Describe the bug**
I happen to have 2 flight control board with same uid (CUAV v5+). According to #10059 , we can change the uuid by changing the MAV_SYS_ID. However, no mater what MAV_SYS_ID I assigned to these two board, they still have the same UUID when I use MAVSDK system.uuid() to read them. 

**Expected behavior**
If I change the MAV_SYS_ID, the UUID should change accordingly based on #10059.

When I use QGC's mavlink inspector->autopilot_version->uid, it returns out to be the same number of what MAVSDK returns to me when I use system.uuid(). The uid2 shows different number though.

bkueng, 2020-01-10:
@julianoes isn't the SDK using uid2?

julianoes, 2020-02-12:
> @julianoes isn't the SDK using uid2?

Not yet unfortunately, see: https://github.com/mavlink/MAVSDK/pull/694

julianoes, 2020-02-12:
Thanks for the issue. I'm closing this but will track it over here: https://github.com/mavlink/MAVSDK/issues/984

closed 2020-02-12 by julianoes