PX4/PX4-Autopilot #13887: mpc: states.acceleration has wrong sign

CarlOlsson, 2020-01-08:
Looking at the acceleration states in the mc position controller it seems to me that they have the wrong sign
https://github.com/PX4/Firmware/blob/92adc5afaa9a1bf8776f8764374fb54ac6b48d38/src/modules/mc_pos_control/mc_pos_control_main.cpp#L453-L453

I.e. they are actually the negative of the acceleration. Is this intended or should I change it?

dagar, 2020-01-08:
> I.e. they are actually the negative of the acceleration. Is this intended or should I change it?

I'm curious as well.

While we're on the subject I also really don't like that the acceleration actually used by the controller isn't logged. (https://github.com/PX4/Firmware/pull/13527)

bresch, 2020-04-08:
Yes, the sign of the acceleration is wrong, but the result is correct, because in the velocity controller, we do `output = vel_error * P + vel_error_int * I + acceleration * D` (normally, we should put the minus sign here)

MaEtUgR, 2020-04-09:
@CarlOlsson Yes, you're completely right in the sense that the sign is wrong for the variable name and that's totally confusing and we should change that. It's only used in the position controller as `vel_dot` https://github.com/PX4/Firmware/blob/0a2c48c69aabbcb9a290098bc324ef1872d87125/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L148 and it's used for the derivative term of the velocity control like an error which requires a sign switch and that's why it's the negative state here.

All correct calculation but totally confusing and we need to immediately change it. I'll open a pr.

MaEtUgR, 2020-04-09:
I changed it here: https://github.com/PX4/Firmware/pull/14629

PR #14629 (2020-04-09): mc_pos_control: correct sign of acceleration state
**Describe problem solved by this pull request**
The sign of the acceleration state passed to the position controller is wrong because it's only used as the derivative of the error in the controller which is opposite sign. The variable should contain what the name says and not the negative value because of how it's used.

fixes #13887

**Describe your solution**
Non-functional change, just change the sign in the correct place to avoid further confusion.

**Test data / coverage**
Untested but I checked it's the only place the variable is used.
closed 2020-04-09 by bresch