PX4/PX4-Autopilot #13889: GPS data not being logged/used

RicardoM17, 2020-01-08:
**Describe the bug**
In one of our test drones, with the current setup (Pixhawk 4 + PX4 master from ~2months ago with really minimal changes) some weird issue is happening with the GPS and flight log/data.

As I'm still not too sure what the issue is or more improtatntly where it comes from the easiest way is to look at some logs. All of them were done today with the same setup/firmware, flying manually in stabilized or Acro. Because of this no performance loss has been observed as we haven't flown in other modes but a test was also made with Position control and everything worked fine.

Log 1 (Bad log):

https://logs.px4.io/plot_app?log=901b0a06-1370-4c37-914f-11cb9f130e2c

Log 2 (Okay log):

https://logs.px4.io/plot_app?log=9f1ae659-5224-46ff-88d0-2948c7d9bc0c

The first log is missing the trajectory map and also it does not have any readings for the Local Position X and Y plots. However it does have GPS  altitude readings as well as the entire plots for GPS Uncertainty and GPS Noise & Jamming plots which seem ok and with normal values.

The second plot however has the map and 3D view of the entire flight despite the fact that local position X and Y, only have relevant data after a certain point which coincides with the following logged messages:

`[ecl/EKF] EKF GPS checks passed (WGS-84 origin set)`
`[ecl/EKF] EKF commencing GPS fusion`

The other GPS plots are like the previous log, i.e. seemingly okay during the entire log.

The expected behaviour would be the flight log is always logged even if for some reason it can't be used for automatic flight due to other reasons like lack of accuracy.

The first approach was to replace the GPS even though this was working fine with the older pixhawk and both GPS present the same issue. 

Random comments:
I noticed in the flight logs the presence of a parameter `EKF2_REQ_GDOP` which is set to the default value of `2.5` but this is a parameter that seems to have been removed. I doubt that this would be the cause however.

The setup is using VESC via UAVCAN. Again I doubt this is the cause.

dagar, 2020-01-08:
In the bad log your vehicle never had a valid global position estimate (vehicle_global_position). 

[image]

At a glance I believe it's because the speed variance reported by your GPS exceeds the default threshold (EKF2_REQ_SACC 0.5).

julianoes, 2020-02-14:
@RicardoM17 did you get to the bottom of  this? Should we close this issue or is there something to be follow up?

RicardoM17, 2020-02-17:
@julianoes Fortunately the issue vanished. Unfortunately we couldn't really see what was the main cause. We did slightly change our setup for other reasons which we will revert soon so I'll keep an eye out to see if it crops up again.

Regarding @dagar comments I'm not so sure if that is the cause. Both logs show similar levels of the speed variance reported by the GPS (though the bad log shows a much larger epv. That being said I would expect that if the EKF checks don't pass that you can't fly in auto modes like Position or mission but that still, even in that scenario, that the GPS/position data gets logged.

dagar, 2020-02-17:
> Regarding @dagar comments I'm not so sure if that is the cause. Both logs show similar levels of the speed variance reported by the GPS (though the bad log shows a much larger epv. That being said I [...]

There are quite a few different checks and they're at different levels. If it's bad enough early on then the estimator could never publish a global position estimate (vehicle_global_position). In other cases you could have a valid global position estimate, but the error (eph, epv) could exceed the failsafe thresholds and commander won't let you enter a position mode.

closed 2020-02-19 by julianoes