PX4/PX4-Autopilot #13892: Pixhawk 3 Pro v4_ failed connection to the RC

Tony3dr, 2020-01-08:
**Describe the bug**
The vehicle did not connect to RC, it would not let the user bind the RC nor calibrate, the same issue was noticed on Master it seems to be working on Stable firmware. This is a brand new flight controller shipped from Drotek, the previous flight controller had bad vibrations and the mag data looked bad. We suspect the issue was something electrical and the flight controller was replaced. Now we are having a failed connection to the RC.  

**To Reproduce**
Steps to reproduce the behavior:
Binding Procedure:

RC Steps:
1- Turn on RC
2- Activate Biding mode (RC will make beeping sounds)

Vehicle Steps:
"With the vehicle turned on"
1- Unplug the receptor
2- Press Binding button and keep pressed
3- Plug in receptor (without letting the binding button go) and make sure both red and green light of the receptor are turned on.
RC Step
1- Disable binding mode
Vehicle Step:
1- Confirm binding by having the receptor with a solid green light on (algo confirm by moving flight modes switches.)

**Log Files and Screenshots**
PR13837: https://review.px4.io/plot_app?log=51c918b7-763c-41c8-b7c8-4a162be0fd3d
Master: https://review.px4.io/plot_app?log=e9b18b85-685e-4f30-8931-9bfdafbc159a
[image]
[image]
[image]

**Drone Information**
- Pixhawk 3 Pro v4
- DJI F450 Frame
- CUAV GPS

**Additional context**
 Below is the last working logs on Master for the Pixhawk Pro. The PixRacer seems to be working fine.
Flight card 1
Log: https://review.px4.io/plot_app?log=35d59e70-c3f1-4bfc-a76a-ec8de20261c3
Flight card 2
Log: https://review.px4.io/plot_app?log=0ebede1a-549f-4638-af23-7dcfe365e371
Flight card 3
Log: https://review.px4.io/plot_app?log=5bbb5e95-acd1-4fd1-a78d-5f2e028d8fb7
Flight card 4
Log: https://review.px4.io/plot_app?log=81e3f2bd-bcca-4263-9b00-20f9beea0d6a

Junkim3DR, 2020-01-15:
### Tested on Pixhawk 3 Pro

**Indoor Test**
Stabilized Flight Mode: Good.

**Note**
Vehicle only flew on Stable firmware, Master still has issues "Not detecting RC".

**Log**
- Stable: https://review.px4.io/plot_app?log=2e26a752-6434-4218-9d58-f2a8f68cc5b3

Tony3dr, 2020-01-16:
@dagar the Pixhawk 3 Pro seems to be working on Stable firmware but not Master. We installed the damaged flight controller on the F450 vehicle and encountered the same issue. Both new and damaged flight controller, vehicle not connecting to RC, it would not let the user bind the RC nor calibrate.

Junkim3DR, 2020-02-17:
### Tested Pixhawk 3 V4Pro

**Stable Firmware**
**Indoor Test**
- Stabilized Flight Mode: Good

**Log**
- Stable Firmware: https://review.px4.io/plot_app?log=6ac7f842-3634-4a68-b233-e415e7169016

**Master Firmware**
**Indoor Test**
- Stabilized Flight Mode: Good

**Log**
- Master Firmware: https://review.px4.io/plot_app?log=f1656fb1-ac5c-4fc0-a406-12ba4f9b3ddd

**Note**
Working as expected on both Stable and Master Firmware.

Tony3dr, 2020-02-24:
The Pixhawk 3 Pro seems to be working on fine on Stable firmware and Master, closing the issue.

closed 2020-02-24 by Tony3dr